RedPitaya/RedPitaya #19: U-Boot patch 0004 is not properly git formatted.

starthal, 2015-03-06:
I wish to apply the patches to u-boot-xlnx so that I can make changes. When I attempt to use `git am` to apply the patchset, git complains "patch format detection failed". Since the patch appears to be the fourth in a set of eleven, it should have the same format as the other patches in the set.

HrRossi, 2015-03-06:
Latest patch count for u-boot in Red Pitaya is 14, and of those, numbers 0004, 0013 and 0014 will not work with 'git am', because of the headers required by git not being there. 
Since git is not used for the patching, I wouldn't see git compatible headers as a requirement per se, but if you prepare a pull request, I'll test and merge it.

starthal, 2015-03-08:
Thanks. I was under the impression that the devs had their own fork of `u-boot-xlnx

starthal, 2015-03-08:
Thanks for the information. I was under the impression that the devs had their own fork of `u-boot-xlnx` and were exporting patches out of the repo to avoid dealing with a subtree. I don't see a reason to make the patches compatible if this isn't the devs' workflow. But I may do it for my own purposes, and if I do I'll submit a pull request.

HrRossi, 2015-03-08:
Now that you mention it, it is probably the case that some have a u-boot fork with RedPitaya-specific changes and generate diffs from that, but - as evidenced by the non git-originated patches - others do not. 
Thanks for contributing.

closed 2015-03-08 by HrRossi