RedPitaya/RedPitaya #23: Please create a tag for version 0.93

pavel-demin, 2015-06-12:
Dear RedPitaya developers,

The version number of the SD image zip file downloadable from redpitaya.com is 0.93.

Looking at the RedPitaya github repository, it's not clear what code is used to build this SD image zip.

Could you please create a tag corresponding to this version?

Kind regards,

Pavel

jeras, 2015-06-12:
The released build is based on our internal git repository. We will again synchronize the public repo with our next release.

noah95, 2015-06-12:
You're talking about a internal repo there, that doesn't really sound like a true open source thing? Why can't you just work on a seperate branch on github? You could only benefit of the contribution by the community.

jeras, 2015-06-12:
There were no functional changes to the FPGA, only fixes for possible clock domain crossing issues with the previous system bus infrastructure.

pavel-demin, 2015-06-12:
Hi Iztok,

Thanks for the information about the status of the latest version.

Maybe I'm too pessimistic, but it's unfortunate to see how the RedPitaya project becomes less and less open:
 - schematics is not open,
 - SDK is not open,
 - latest FPGA code is not open.

Since the code is not in this repository, I'm closing this ticket as irrelevant.

Cheers,

Pavel

closed 2015-06-12 by pavel-demin

pavel-demin, 2015-06-13:
For information.

I've created a new thread of the RedPitaya forum to discuss the release 0.93 and GPLv2 compliance:

http://forum.redpitaya.com/viewtopic.php?f=8&t=559