RedPitaya/RedPitaya #24: rp_data_get_signals() in ngx_http_rp_module assumes fixed signal count and size

aAXEe, 2015-06-22:
Hi all,
I am writing a custom web application. Changing the number and size of the signals works not as expected because the ngix module assumes fixed values:
https://github.com/RedPitaya/RedPitaya/blob/master/Bazaar/nginx/ngx_ext_modules/ngx_http_rp_module/src/rp_data_cmd.c#L413
```C
if(rp_signals == NULL) {
int i;
rp_signals = (float **)malloc(3 * sizeof(float *));
for(i = 0; i < 3; i++) {
rp_signals[i] = (float *)malloc(2048 * sizeof(float));
}
}
```

Especially if you have bigger values in your app bad things might happen.

RedPitaya, 2015-06-22:
Hello sir,Thank you for notifying. We will take appropriate action and fix the problem.If there is anything else you'd find please notify us asap.Kind regards Martin Cimermanredpitaya.com                                                                                                                                                                                                                From: aAXEeSent: Monday, June 22, 2015 09:56To: RedPitaya/RedPitayaReply To: RedPitaya/RedPitayaSubject: [RedPitaya] rp_data_get_signals() in ngx_http_rp_module assumes fixed signal count and size (#24)Hi all,
I am writing a custom web application. Changing the number and size of the signals works not as expected because the ngix module assumes fixed values:
https://github.com/RedPitaya/RedPitaya/blob/master/Bazaar/nginx/ngx_ext_modules/ngx_http_rp_module/src/rp_data_cmd.c#L413

if(rp_signals == NULL) {
int i;
rp_signals = (float **)malloc(3 * sizeof(float *));
for(i = 0; i < 3; i++) {
rp_signals[i] = (float *)malloc(2048 * sizeof(float));
}
}

Especially if you have bigger values in your app bad things might happen.

HrRossi, 2015-06-23:
I would suggest to leave the existing interface as it is and add a function to the bazaar-interface to transfer custom json data in both directions. Something like
```c
int customTransfer(cJSON *input, cJSON *output);
```
Let the web-app define its own data format and let it do its own parsing. With just the additional contract that all incoming cJSON structures are to be allocated and freed by the bazaar and all outgoings are allocated by the app and freed by the bazaar

b-swat, 2023-07-03:
This mechanism for working with signals is no longer used. Need to use CCustomSignal<T>

closed 2023-07-03 by b-swat