RedPitaya/RedPitaya #28: add USB ID 0bda:8179

andras, 2015-07-31:
Hi,

I have a TP-Link TL-WN725N v2, which uses the RTL8188EUS chipset and should therefore work out of the box, but it doesn't because its USB ID 0bda:8179 is unknown to the RedPitaya ecosystem.

Could you please add the required entries to OS/linux/patches/add_rtl8192cu.patch and upload a new ecosystem build?  I haven't got the time just now to set up the build environment required for doing this.

Thanks!

jeras, 2015-07-31:
We are working on a new release, two wireless chips will be supported, I am not sure which ones right now. The kernel will be 3.19, with some wireless driver patches from Raspberry PI. Please update this bug report after the new release will be available.

jeras, 2015-10-18:
Could you please check if this is working for you in version 0.94.

closed 2023-07-03 by b-swat