RedPitaya/RedPitaya #29: Two axi_master with same ID

DF4IAH, 2015-09-10:
In the file FPGA/release1/fpga/code/rtl/red_pitaya_ps.v file there is mentioned that the same ID for two differtent axi_master is not OK.

I solved that issue with re-factoring the two different axi0_master and axi1_master to a new naming convention axi_master[1:0]. With that help a generate can be done on the instantiation of the axi_master.
PR follows ...

Greetings,
  Uli

HrRossi, 2015-09-11:
Seeing as you already created a fix for the matter, I am afraid to say that this is a non-issue. 

The mentioned TODO is based on a misconception. Internally, the ID is padded to 13bit with values that are different for each master, so there is never any overlap in IDs between different masters.

HrRossi, 2015-09-11:
By the way, I didn't know that you even could declare arrays of instances. So, thanks for showing me some new syntax ;O)

closed 2015-09-11 by HrRossi