RedPitaya/RedPitaya #32: Nginx Makefile missing

jeanminet, 2015-09-25:
Reference commit : 7d0769e6b77fcf66ecca79e9b667e8fed90ac1e9

I do not find the Makefile required to compile Nginx:

```make
$(NGINX): $(URAMDISK) $(LIBREDPITAYA) $(WEBSOCKETPP_DIR) $(CRYPTOPP_DIR) $(LIBJSON_DIR) $(LUANGINX_DIR) $(NGINX_SRC_DIR) $(BOOST_DIR)
	$(MAKE) -C $(NGINX_DIR) SYSROOT=$(SYSROOT)
	$(MAKE) -C $(NGINX_DIR) install DESTDIR=$(abspath $(INSTALL_DIR))
```

Could you please add it to the repository ?
Thank you !

RedPitaya, 2015-09-26:
Hi Jeanminet,
this part of the code was not yet ready to be published, but we will sure update it soon.
At the moment we just releasing new website and new OS  with many new features,
so we need all our development resources for testing and fixing most critical bugs.
Thank you for your understanding.
kind regards, Red Pitaya team

jeras, 2015-10-18:
We are almost ready to release a new version containing everything except payable applications. It will be possible to build everything into a functional whole, except for the mentioned applications.

jeras, 2015-10-19:
The sources are now published.

closed 2015-10-19 by jeras