RedPitaya/RedPitaya #33: Error in SCPI BIN mode header

hadmack, 2015-10-01:
In `scpi-server/3rdparty/libs/scpi-parser/libscpi/src/parser.c`:

line 516 incorrectly:
`result += writeBinHeader(context, size, sizeof(float));`

should be:
`result += writeBinHeader(context, size, sizeof(int16_t));`

jeras, 2015-10-01:
Thanks.

We are currently migrating from scpi-parser v1.2 to v2.1, where we should use the builtin arbitrary data header. We are also working on automated SCPI unit tests, it will take us some time.

For now I will commit this change to an internal branch.

hadmack, 2015-10-01:
Hopefully the switch to v2.1 will take care of things.  I am not having much success with the BIN transfer mode at present.  In ASCII mode my data looks fine but in BIN mode the low byte of negative values is always 0x00 but it is correct for positive values.  The resulting waveform is a staircase for negative values and smooth for positive.  I did not have much luck debugging this but I suspect that the htons() conversion is mangling the data somehow.

jeras, 2015-10-18:
Yes, it was htons(), we managed to fix this issue by adding proper type casting. We will hopefully release this fix soon.

closed 2015-10-18 by jeras