RedPitaya/RedPitaya #37: missing /opt/redpitaya/hostapd.conf

pavel-demin, 2015-10-26:
Hi Iztok,

I've just had a look at the new Red Pitaya SD card image trying to understand why Wi-Fi access point does not work (there is even a [forum thread](http://forum.redpitaya.com/viewtopic.php?f=8&t=794) dedicated to this problem).

Apparently, [line 62 in debian.sh](https://github.com/RedPitaya/RedPitaya/blob/master/OS/debian/debian.sh#L62) does not work and /opt/redpitaya/hostapd.conf is not copied to the SD card image.

As a workaround, I tried to download this file from GitHub with the following commands:
```
/opt/redpitaya/sbin/rw
curl -L https://raw.githubusercontent.com/RedPitaya/RedPitaya/master/OS/debian/overlay/etc/hostapd/hostapd.conf -o /opt/redpitaya/hostapd.conf
/opt/redpitaya/sbin/ro
```
After a reboot, Wi-Fi access point started to work.

Could you please fix debian.sh?

Cheers,

Pavel

pavel-demin, 2015-10-26:
Another related problem.

The `PATH_REDPITAYA` variable used in `/opt/redpitaya/sbin/ro` and `/opt/redpitaya/sbin/rw`  is not defined when these scripts are called from `/etc/default/hostapd`.

Adding `export PATH_REDPITAYA=/opt/redpitaya` in the very beginning and `unset  PATH_REDPITAYA` in the very end of /etc/default/hostapd solves this problem.

RedPitaya, 2015-10-27:
Regarding the first issue. The problem is not with the debian.sh script, but with me making a shortcut during the SD card Image build process. Since the full SD card image build process takes a long time, I usually just used an existing Debian image and just overwrite the FAT partition with the latest ecosystem ZIP. I will be carefull with the next RC to also add this missing file manually.

For the second step I will do something similar to your recommendation.

RedPitaya, 2015-10-29:
Hi Pavel, thanks for the detailed bug report and recommended fixes. This is now Fixed in RC17, which is now also linked as the latest version. Sources were also pushed to Github.

pavel-demin, 2015-10-29:
Thanks for the fixes.
I've just checked the latest commits and the RC17 SD card image.

The /opt/redpitaya/hostapd.conf is now in the RC17 SD card image.

However I don't see the changes corresponding to https://github.com/RedPitaya/RedPitaya/commit/a9c2ec0c6357d926eb8dd5f87630de933b9fd952  in the RC17 SD card image. It still contains a previous version of /etc/default/hostapd.

RedPitaya, 2015-10-29:
same mistake as before, I used the ext4 partition from the previous image, will create a new RC soon

pavel-demin, 2015-10-29:
Just tested new /etc/default/hostapd and found a new problem:
```
    iw wlan0 info > /dev/null 2>&1
    PATH_REDPITAYA=/opt/redpitaya /opt/redpitaya/sbin/rw
    if [ $? -eq 0 ]
```
should be replaced with
```
    PATH_REDPITAYA=/opt/redpitaya /opt/redpitaya/sbin/rw
    iw wlan0 info > /dev/null 2>&1
    if [ $? -eq 0 ]
```
otherwise, `if [ $? -eq 0 ]` can't check the exit code of the iw command.

jeras, 2015-11-01:
this is now ficed in v0.94-RC19

closed 2015-11-01 by jeras