RedPitaya/RedPitaya #38: IST RealProbe Flow & Oscilloscope 0.94-400 does not work

pavel-demin, 2015-10-26:
Just tested all the Bazaar applications with the latest SD card image (red_pitaya_OS_v0.94-RC16_21-Oct-2015.img.zip). All seem to work except "IST RealProbe Flow & Oscilloscope 0.94-400".

The error message is "Could not start the application".

I tried to uninstall/install it a couple of times but it didn't help.

RedPitaya, 2015-10-27:
Regarding IST, this was a contributed application, and we did not spend much time making it work, we also do not know how to check if it works properly. In any case all applications had to be recompiled for armhf. The other change was the handling of FPGA bitstreams, now there is a fpga.conf file containing the path to the appropriate FPGA bitstream. In release 0.93 all applications used the same bitstream, now new applications use a newer bitstream.

Regarding oscilloscope, I usually focus my tests on gen+scope, I will check what went wrong with oscilloscope. From the error it seems to be something wrong with `controllerhf.so`.

pavel-demin, 2015-10-27:
Oscilloscope is OK.

It's just one application called "IST RealProbe Flow & Oscilloscope 0.94-400" that does not work.

RedPitaya, 2015-10-27:
Let me describe my debug process first, before I store it into a readme. All steps are performed over SSH on a Pitaya board:

```bash
sudo apt-get install strace
systemctl stop redpitaya_nginx
rw
vi /opt/redpitaya/www/conf/nginx.conf
```
here change the line:
```
daemon off;
```
into:
```
daemon off;
master_process off;
```
This will allow running Nginx in some debug modes:
```
LD_LIBRARY_PATH=/opt/redpitaya/lib strace /opt/redpitaya/sbin/nginx -p /opt/redpitaya/www/
```
After running the application I get:
```
clone(child_stack=0x33cb2f68, flags=CLONE_VM|CLONE_FS|CLONE_FILES|CLONE_SIGHAND|CLONE_THREAD|CLONE_SYSVSEM|CLONE_SETTLS|CLONE_PARENT_SETTID|CLONE_CHILD_CLEARTID, parent_tidptr=0x33cb3498, tls=0x33cb38f0, child_tidptr=0x33cb3498) = 1832
open("/dev/mem", O_RDWR|O_SYNC <ptrace(SYSCALL):No such process>
+++ killed by SIGSEGV +++
Segmentation fault
```

A segmentation fault would probably be easy to debug using `gdb`, but here it would probably help to have all code compiled natively. Our Makefile is not in a shape yet, which would us allow to compile our Nginx server on Red Pitaya, mostly due to Buildroot dependencies.

I tried to run Nginx in `gdb`, but it fails early with:
```
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Cannot access memory at address 0x0

Program received signal SIGILL, Illegal instruction.
0x36dd4da8 in ?? () from /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.0.0
```
Once I read some recommendations on how to avoid this one and it involves Kernel changes, which we tried once, but did not get far.

I will not debug this further now.

closed 2023-07-03 by b-swat