RedPitaya/RedPitaya #42: possibility to configure a default IP address if there is no DHCP server

pavel-demin, 2015-11-27:
Hello,

Would it be possible to add possibility to configure a default IP address if there is no DHCP server as it was done in the earlier versions of the Red Pitaya ecosystem and as it's described on the following wiki page:
http://wiki.redpitaya.com/index.php?title=User_Manual#Network_DHCP_configuration

This functionality was already requested several times on the Red Pitaya forum:
http://forum.redpitaya.com/viewtopic.php?f=7&t=568#p2665
http://forum.redpitaya.com/viewtopic.php?t=1040&p=3582#p3582
http://forum.redpitaya.com/viewtopic.php?t=1036&p=3581#p3581

To configure a default IP address if there is no DHCP server, the following lines should be added to `/etc/dhcp/dhclient.conf` on the Debian based system:
```
timeout 20;

lease {
  interface "eth0";
  fixed-address 192.168.1.100;
  option subnet-mask 255.255.255.0;
  renew 2 2030/1/1 00:00:01;
  rebind 2 2030/1/1 00:00:01;
  expire 2 2030/1/1 00:00:01;
}
```

Best regards,

Pavel

jeras, 2015-11-27:
Thanks Pavel, I was ignoring this one far too long. I committed the change to our internal repo. Will test it and publish a new RC next week.

jeras, 2015-12-02:
This has now been merged into RC21, I tried it once and it seems to work.

closed 2015-12-02 by jeras