RedPitaya/RedPitaya #43: Different interpretation of channel value in RP_GenTrigger(), RP_ParseChArgv(), rp_GenTrigger() and gen_Trigger()

uiae, 2015-11-28:
I think I might have spotted a some bugs in the scpi-server application. They occur when setting the SOUR#:TRIG:IMM value.

As this is more an architectural/consistency kind of problem, I am unable write a patch resolving the issue.

### 1) RP_ParseChArgv allows a larger channel range than rp_channel_t can hold

When submitting the above command, we land in the following function (scpi-server/generate.c)
```
scpi_result_t RP_GenTrigger(scpi_t *context) {
    
    rp_channel_t channel;
    int result;

    if (RP_ParseChArgv(context, &channel) != RP_OK){
        return SCPI_RES_ERR;
    }
   ....
````
It parses the channel number with the help of `RP_ParseChArgv` (scpi-server/common.c) and assigns the parsed channel number to channel, which is a rp_channel_t enum. The problem is that `RP_ParseChArgv` checks whether the channel number is within the interval [0;3]. (MIN_CH==0 and MAX_CH==3). However, the rp_channel_t enum can only hold two values (RP_CH_1 and RP_CH_2). Moreover, `RP_ParseChArgv` subtracts 1 _AFTER_ checking the range meaning that sending the command `SOUR0:TRIG:IMM` yields a channel number of -1.

```
int RP_ParseChArgv(scpi_t *context, rp_channel_t *channel){
  if(ch_usr[0] < MIN_CH || ch_usr[0] > MAX_CH){
    RP_LOG(LOG_ERR, "ERROR: Invalid channel number: %.*s\n", 50, context->param_list.cmd_raw.data);
  return RP_EOOR;
}
```
```
typedef enum {
    RP_CH_1, //!< Channel A
    RP_CH_2  //!< Channel B
} rp_channel_t;
```

### 2) The same value is interpreted as channel number and then as mask without proper conversion
According to the SCPI-Command-Documentation (.odt), the rp_channel_t enum and the variable names, the SCPI-Command as well as the scpi-command-handling functions assume that the supplied number corresponds a channel number. But the function `rp_GenTrigger(int mask)`, to which this value is passed handles this value as a mask rather than a channel number. This can be seen when following the call path from `RP_GenTrigger(scpi_t *context) -> rp_GenTrigger(int mask) -> gen_Trigger(int mask)`:

In RP_GenTrigger (scpi-server/generate.c), the supplied value is assumed to be a channel number:
```
scpi_result_t RP_GenTrigger(scpi_t *context) {
    
    rp_channel_t channel;
    int result;

    if (RP_ParseChArgv(context, &channel) != RP_OK){
        return SCPI_RES_ERR;
    }

    result = rp_GenTrigger(channel);
    if(result != RP_OK){
        RP_LOG(LOG_ERR, "*SOUR#:TRIG:IMM Failed to set immediate "
            "trigger: %s\n", rp_GetError(result));
        return SCPI_RES_ERR;
    }

    RP_LOG(LOG_INFO, "*SOUR#:TRIG:IMM Successfully set immediate trigger.\n");
    return SCPI_RES_OK;
}
```

But in the following functions (rpbase/rp.c and rpbase/gen_handler.c) the same number is interpreted as mask rather than a channel number:
```
int rp_GenTrigger(int mask) {
    return gen_Trigger(mask);
}
```
```
int gen_Trigger(int mask) {
    switch (mask) {
        case 1:
            ECHECK(gen_setGenMode(RP_CH_1, RP_GEN_MODE_BURST));
            return generate_setTriggerSource(RP_CH_1, RP_GEN_TRIG_SRC_INTERNAL);
        case 2:
            ECHECK(gen_setGenMode(RP_CH_2, RP_GEN_MODE_BURST));
            return generate_setTriggerSource(RP_CH_2, RP_GEN_TRIG_SRC_INTERNAL);
        case 3:
            ECHECK(gen_setGenMode(RP_CH_1, RP_GEN_MODE_BURST));
            ECHECK(gen_setGenMode(RP_CH_2, RP_GEN_MODE_BURST));
            return generate_simultaneousTrigger();
        default:
            return RP_EOOR;
    }
}
```

### 3) Different offset to encode the channels
`gen_Trigger(int mask)` assumes channels to be encoded as follows:
```
channel 1:     1
channel 2:     2
both channels: 3
```

The parse function `RP_ParseChArgv` as well as the `rp_channel_t`-enum encode channels as follows:
```
channel 1: 0
channel 2: 1
```
This leads to the problem that if you would actually want to trigger channel1 (as defined by `rp_channel_t` and parsed by `RP_ParsechArgv` you would have to send `SOUR2:TRIG:IMM` instead.

jeras, 2015-11-30:
Thanks for the detailed description.

I could not prepare a consistent fix, which would be both logical and would keep existing functionality.

First regarding channel indexes. On the outside (based on what is common practice) the channels are enumerated as `1`, `2`. Internally channels are enumerated from `0`, since this makes sense in the C language. There for the `-1` in `RP_ParseChArgv`.

So I moved the channel indexing in the SOUR#:TRIG:IMM command to behave as:
`SOUR1:TRIG:IMM` will trigger channel 1 (internal index is 0)
`SOUR2:TRIG:IMM` will trigger channel 2 (internal index is 1)
`SOUR3:TRIG:IMM` will trigger channel 1, 2 (internal index is 0, 1)
`SOUR4:TRIG:IMM` will trigger channel 1, 2 (internal index is 0, 1)

This also means allows channel numbers 3, 4, which do not exist, but are only used for this combined trigger.

I am writing a simpler and more generic API, and FPGA RTL, which should avoid many issues of this kind. But there were other priorities.

uiae, 2015-11-30:
Thanks for the clarification!
I didn't know, that specifying source 3 and 4 should trigger both channels. It now makes sense. ;)

However, to my knowledge, the current master branch's HEAD does not implement the behaviour which you described above.
Because:
`SOUR1/CH1` and `SOUR2/CH2` are parsed as channel 0 and 1 respectively
But:
`genTrigger(int mask)` uses 1 for triggering `CH1`, 2 for triggering `CH2` and 3 for both channels.

Therefore, supplying `SOUR1:TRIG:IMM` will cause `genTrigger` to hit the default case in the switch statement (which throws an error).
And supplying `SOUR2:TRIG:IMM` will cause `genTrigger` to hit the 1 case (which actually triggers `CH1`)

Regarding the `-1` in `RP_ParseChArgv`: Declaring CH1 as 1 in the API but storing it internally as 0 makes sense (default c enums start at 0 anyways). The problem I see here is that the range check (CH_MIN<= value =>CH_MAX) works on the internal channel representation (values 0-3) but the check is issued on values 1-4 which are not yet converted to the internal representation. Redefining CH_MIN=1 and CH_MAX=4 would fix this I guess.

It is great to hear that the API is still in active development and constantly improving. Thanks for the effort and the great work! :)

jeras, 2015-12-02:
Actually I didn't know either, that specifying source 3 and 4 should trigger both channels. This is a dirty hack due to a poorly designed interface. I tried to keep as much existing functionality as possible.

The changes are now included in RC21 (you can have a look at the changelog), but I did not test them yet, so I will wait till someone runs a few tests before closing this issue.

uiae, 2015-12-21:
I finally had time to review the commit you are mentioning. Everything seems alright now! Thanks you jeras the quick handling of the bug!

closed 2023-07-03 by b-swat