RedPitaya/RedPitaya #47: readme clarification

baj12, 2015-12-15:
Hi,

could you please elaborate on the following that is written in the README.md:

Xilinx Vivado 2015.2 FPGA development tools. The SDK (bare metal toolchain) must also be installed, be careful during the install process to select it. Preferably use the default install location.

It is not clear to me what to download, if it is really necessary to register there, etc. I am having difficulties finding FPGA and also have no clue what metal toolchain refers to. I hope it is not necessary for me to learn everything about vivado, because I rather concentrate on the pitaya...

Thanks for the clarification,

Bernd

jeras, 2015-12-15:
Hi Bernd,

The Vivado software is needed to compile the FPGA Verilog code. The bare metal toolchain is for compiling applications which do not run on top of linux, but instead run without an OS providing a posix API and virtualized memory. This applications are bootloaders like FSBL and U-Boot. Yes, it is necessary to register if you wish to download and install the Vivado software.

It is possible to compile only Applications running on Linux (api, apps-free, scpi, ...), for this Vivado and the bare metal toolchain are not needed. It is up to the user to understand the Makefile. It is also possible to compile applications directly on the Debian system on the Red Pitaya board, for example `Examples/C/`.

Regards,
Iztok

baj12, 2015-12-15:
Hi Iztok,

thanks for the quick responds.

Specifically, when I go to the download page (http://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/vivado-design-tools/2015-2.html) do I need version 2015.2 or would it be better to use the latest one (2015.4).

Within either what should I download/install. There are several downloads available. (Vivado Web Install Client; Vivado Design Suite; Vivado Lab Edition; Vivado Hardware Server, (none of which mentions FPGA...)

What about the license manager? which license option is recommended?

I still don't know where there could be a choice for selecting the following: "The SDK (bare metal toolchain) must also be installed, be careful during the install process to select it."

In the end I want to develop a tool that stores measurements for a given period of time once triggered by a button.

Thx again, and sorry for being a bit slow...

Best,
Bernd

jeras, 2015-12-15:
Hi,

There are some source files, which require a specific version of the Vivado
tool, and currently this version is 2015.2. I usually download "Vivado
Design Suite" which allows for a fully offline install process, otherwise
"Vivado Web Install Client" should probably be OK too, it should be able to
download less files. Mentioning FPGA in Xilinx software names would be like
mentioning water in sea life documentaries, everything from Xilinx is about
FPGA. The option for installing the SDK is about the third window:
1. license agreement
2. edition options, choose the design edition
3. enable components, DSK, ZYNQ-7000, must be enabled, Virtex and some
others could be disable to save space.

For your tool, our current FPGA code limits data acquisition to 16384
samples, which limits the 'given period of time' depending on the chosen
decimation.

Regards,
Iztok Jeras

baj12, 2015-12-15:
Iztok, thanks a lot!!! that is helping!
just a general question about the 16384 samples: is this a final limit? or can one change this in any way/future? What would be the reason for this limitation?
I will have to play much more with the system, but I am sure I will find a way of averaging if this limit proves to be hard...

Thanks again,

Bernd

jeras, 2015-12-15:
With FPGA changes it would be possible to increase the buffer size to about
a double, but the value is currently hardcoded in many driver/API files so
it would require many changes to actually work. We are also working on a
DMA which would be able to store up to 128Msamples (mega) in the main DDR3
memory (half reserved for data storage, the rest for the OS).

baj12, 2015-12-15:
thanks a lot.
B

closed 2015-12-15 by baj12