RedPitaya/RedPitaya #56: Buffer overflow in scpi's RP_AcqSamplingRateHzQ()

HrRossi, 2016-02-20:
In scpi-server/src/acquire.c RP_AcqSamplingRateHzQ() the return value (usually "125000000 Hz") is printf'ed into a char[1] buffer. 

This leads to the value being returned without linefeed and immediate closing of the connection.

jerzydziewierz, 2016-04-21:
I just got the same issue,
typing in terminal:
========
ACQ:SRAT?
125000000 HzConnection closed by foreign host.
==============

DF4IAH, 2016-04-30:
No one of the Collaborators to change that? Would this snippet help?
Is a "\n" needed for the result string? I have low expirience for SCPI by now.

```
diff --git a/scpi-server/src/acquire.c b/scpi-server/src/acquire.c
index 1a92a2d..c608317 100644
--- a/scpi-server/src/acquire.c
+++ b/scpi-server/src/acquire.c
@@ -234,8 +234,9 @@ scpi_result_t RP_AcqSamplingRateHzQ(scpi_t *context) {
     }
 
     // Return back string result
-    char samplingRateString;
-    sprintf(&samplingRateString, "%.0f Hz", samplingRate);
+    size_t str_len = 16;
+    char samplingRateString[str_len];
+    snprintf(&samplingRateString, "%.0f Hz", samplingRate);
 
     //Return string in form "<Value> Hz"
     SCPI_ResultMnemonic(context, &samplingRateString);

```

HrRossi, 2016-05-02:
A while ago, the RedPitayans [commented](https://github.com/RedPitaya/RedPitaya/pull/52#issuecomment-180544718) on a PR that they were reworking the API. Seeing how closely the API and the SCPI server are intertwined, I'm holding off on changes in that area.

sanpii, 2017-04-30:
I didn’t see this issue before, I made a PR https://github.com/RedPitaya/RedPitaya/pull/110

jeras, 2017-05-03:
Hi sanpii,

This pull seemed obvious enough to merge regardless od our future API changes.

But for larger changes I will have trouble reviewing them, since it has been some time since I touched the SCPI code. And I only plan to get back to it, for major API changes.

We now have a cleaner API, which is still written in Python. The DMA code used to store samples in main RAM is still missing, I hope we get the time to write it in the next few months.

Regards,
Iztok Jeras

sanpii, 2017-05-03:
> But for larger changes I will have trouble reviewing them, since it has been some time since I touched the SCPI code. And I only plan to get back to it, for major API changes.

I understand your point of view.

> We now have a cleaner API, which is still written in Python. The DMA code used to store samples in main RAM is still missing, I hope we get the time to write it in the next few months.

Is there in the `api2` directory?

jeras, 2017-05-03:
The `api2` directory contains the current state of the DMA code written in C.
The code is limited to a single DMA and is only used by the Logic Analyzer application.
The other parts of the `api2` code are mostly unused experiments, although they helped me to debug generator burst mode.

The Ptthon API for the latest FPGA code `mercury` is here:
https://github.com/RedPitaya/jupyter/tree/master/redpitaya
But as I said DMA code is still missing.

closed 2023-07-03 by b-swat