RedPitaya/RedPitaya #67: Please create a tag for version 0.95

pavel-demin, 2016-05-16:
Dear RedPitaya developers,

The version number of the SD image zip file downloadable from redpitaya.com is 0.95.

Looking at the RedPitaya github repository, it's not clear what code is used to build this SD image zip.

Could you please create a tag corresponding to this version?

Kind regards,

Pavel

HrRossi, 2016-05-17:
Seconded. It's been a while that 0.95 has been released for binary download, and clearly it's not been built from code in this repository.

jeanminet, 2016-05-18:
Red Pitaya developers responded to the same question in the forum (http://forum.redpitaya.com/viewtopic.php?f=8&t=1451)

> Hi guys
Yes, GitHub repo. is not yet updated to 0.95 version.
It will be updated in month or less.
Sorry for the inconvenience.
All Best, RP

HrRossi, 2016-05-18:
I realize that this information is helpful for anybody who might chance upon this conversation, but the fact that you quote this to the two most active users on that same forum still made me smile ;O)

DF4IAH, 2016-05-18:
Thanks for information. This is number **three** of active developpers. May I introduce myself?   :-)

**BTW**: why are adding PR requests for the bazaar silently ignored? RadioBox is ready to
be added.

All the best also from me
  Uli, DF4IAH.

jeras, 2016-05-19:
Hi, I was (am, will be) on a study vacation so the release procedure was a bit improvised by another coworker. I will try to prepare the sources for a release in the next few days.

pavel-demin, 2016-05-20:
Thanks a lot for creating the [v0.95-1](https://github.com/RedPitaya/RedPitaya/tree/v0.95-1) tag!

closed 2016-05-20 by pavel-demin

jeras, 2016-05-20:
Hi,

Sources are now available in a new branch:
`release-v0.95`
The tagged version will not compile properly, but with the latest commit on the branch compiling should work properly.

DF4IAH, 2016-05-21:
Thank you very much!