RedPitaya/RedPitaya #69: Is there a reason for branching 0.95 from 0.94 ca. RC22 ?

HrRossi, 2016-05-21:
I'd like to know if there are known problems with the release candidates 0.94 >RC22 that made you base the 0.95 release on RC22. 

There have been several bugfixes and enhancements to master after RC22 that are missing from 0.95 . For example the cores are now limited to 500MHz again, tool dependency is back to Vivado 2015.2, and the AXI acquisition logic is broken with the 0.95 "stable" image. 

Regards
Nils

jeras, 2016-05-21:
Hi Nils,

The main reason was a problem with the network setup sequence. Systemd
reported the network as ready too early, so discovery was started before
the network was ready, and failed.

The next release will be based on Vivado 2015.4 or 2016.1 (depending on how
we manage to solve some issues with kernel 4.4), with the CPU clocking
fixed.

AXI acquisition logic and SW will not be maintained, we prefer to use the
DMA from Xilinx, which has a Linux kernel driver available.

Regards,
Iztok

HrRossi, 2016-05-21:
Thanks for clarifying. 

> AXI acquisition logic and SW will not be maintained, we prefer to use the DMA from Xilinx, which has a Linux kernel driver available.

Oh, so you are working on connecting the ADC to a streamwriter or similar in the block design ? I'd like to hear more about that, if possible. 

Regards, 
Nils

DF4IAH, 2016-05-21:
... makes me somehow sad having improved the system (CPU 666 MHz) and then all is vanished, again. That frustrades much.

Do you understand, why so many open-source developers **_went away_** from Red Pitaya?
It is exactly that fact, that the main development is made beside of the GitHub and the **synchronization**
between Red Pitaya Headquaters and the GitHub **is not done properly**. Well, what can I say to that?

How can be GitHub placed as the center of development? Of course, you can make your own packets that are closed source. But keep in mind, I bought this device because of an open source development, as many more known friends of me. Now what? In the meantime it is easy to grab for cheap V1.1 hardware at eBay and others. What do that mean about the user satisfaction? Any ideas?

Hopefully your management does find a more customer friendly solution.

I like the concept of Red Pitaya very very much. What results in frustration is the "open source", that is not really lived as open source by Red Pitaya, as it is defined by the open source community. Please feel free to uplift the user experience with your device(s).

Regards,
  Uli

jeras, 2016-05-21:
Hi Ulrich,

You asked some questions, I just can not discuss publicly.

Our problems with accepting pull requests are as annoying to us as are to you, and although a fix might seem trivial to you, it takes time, effort and persuasion. I might not make it a priority, but I will definitely spend time finding and implementing a solution.

Regards,
Iztok

closed 2023-07-03 by b-swat