RedPitaya/RedPitaya #70: Can't build RedPitaya

Split7fire, 2016-05-30:
OS: Archlinux x86_64

> gdate.c:2497:7: error: format not a string literal, format string not checked [-Werror=format-nonliteral]
>        tmplen = strftime (tmpbuf, tmpbufsize, locale_format, &tm);

in >>> host-libglib2 2.38.2 Building step

HrRossi, 2016-05-30:
What version of gcc is on your system? There are several issues when compiling with a host gcc version >= 5, [see here](http://forum.redpitaya.com/viewtopic.php?p=4633#p4633) for possible workarounds.

Split7fire, 2016-05-31:
gcc is 6.1.1, ok, i'll see

Split7fire, 2016-05-31:
I'm still can't build os/buildroot
What i tried so far:
1. Patches for cmake & nginx from above post
2. Installed gcc 4.9.3 (and using "make HOSTCC=gcc-4.9 HOSTCPP=cpp-4.9 HOSTCXX=g++-4.9
")

P.S. Also it is strange to that redpitaya closely tied to Xilinx SDK version (pre-latest version), but tools to build it quite old (gcc 4.9 released in april 2014 & Ubuntu 14.04 is old-stable). Can you clarify it?

jeras, 2016-05-31:
In our README.md we currently recommend a Linaro compiler 2014.11, the main reason is that Xilinx only provided a compiler for the armEL architecture, while armHF is a much better choice. Since Vivado 2015.4 Xilinx also provides an armHF compiler, which will probably be used in future releases. Note the compiler provided by Xilinx is even older then our recommendation, it is based on Linaro 2014.9.

Split7fire, 2016-06-03:
To sum it up:
* you can't build redpitaya on something newer than gcc 4.9, Vivado 2015.4 and Linaro 2014.11
* for those who use rolling-based Linux distributions, there is no way except virtual machine with (creepy) Ubuntu 14.04
Am I right?

closed 2023-07-03 by b-swat