RedPitaya/RedPitaya #77: can't configure FPGA clock frequency in v0.96

pavel-demin, 2016-08-29:
Dear Red Pitaya developers,

I've just tested the latest stable SD card image (red_pitaya_OS-v0.96-RC1-20-14_jul.img.zip) and found that the following device isn't available anymore:
```
/sys/devices/soc0/amba/f8007000.devcfg/fclk
```

I've also found that `f8007000.devcfg` is renamed to `amba:fpga_bus@0/f8007000.devcfg` with the following patch:
https://github.com/RedPitaya/RedPitaya/blob/master/patches/zynq-7000.patch

Unfortunately, `amba:fpga_bus@0/f8007000.devcfg` doesn't have any `fclk` entry.

My SDR applications require `fclk` to configure the FPGA clock with a slightly higher frequency (143 MHz).

Could you please add the `fclk` entries to one of the future version of the default Red Pitaya device tree.

If I'm not mistaken the following lines in zynq-7000.patch
```
  clocks = <&clkc 12>;
  clock-names = "ref_clk";
```
should be replaced with
```
  clocks = <&clkc 12>, <&clkc 15>, <&clkc 16>, <&clkc 17>, <&clkc 18>;
  clock-names = "ref_clk", "fclk0", "fclk1", "fclk2", "fclk3";
```
Best regards,

Pavel

jeras, 2016-09-05:
Hi Pavel,

We are thinking of removing this patch, at least for now. We should be able to test and upload it today.

Sorry for the late reply. I was not at work last week, had other more important stuff to do.

pavel-demin, 2016-09-05:
Hi Iztok,

Thanks for the news. Another solution could be to set the fclk0 to 143 MHz by default. As far as I know, it should be OK for all other Red Pitaya applications.

Best regards,

Pavel

jeras, 2016-09-05:
We decided to revert this change, updated sources are already available on github. I am not sure when will we publish a new release with this change included, probably next week.

closed 2023-07-03 by b-swat