RedPitaya/RedPitaya #80: Issue with OS/debian/build.sh, no space left on device

andrewhodel, 2016-10-14:
When I run it here's the output:

```
Setting up libcunit1-ncurses-dev (2.1-3-dfsg-2) ...
Processing triggers for libc-bin (2.23-0ubuntu3) ...
root@ThinkPad-11e:/# 
root@ThinkPad-11e:/# exit
'OS/debian/overlay/etc/profile.d/profile.sh' -> 'root/etc/profile.d/profile.sh'
'OS/debian/overlay/etc/profile.d/alias.sh' -> 'root/etc/profile.d/alias.sh'
'OS/debian/overlay/etc/profile.d/redpitaya.sh' -> 'root/etc/profile.d/redpitaya.sh'
root@ThinkPad-11e:/# echo root:root | chpasswd
root@ThinkPad-11e:/# apt-get clean
root@ThinkPad-11e:/# history -c
root@ThinkPad-11e:/# exit
cat: write error: No space left on device
```

Also strangely, the images which are being downloaded are named differently and have different sizes.

```
-rw-r--r--  1 root root 3.5G Oct 14 11:47 redpitaya_ubuntu_11-44-42_14-Oct-2016.img
-rw-r--r--  1 root root  44M Oct 14 11:47 redpitaya_ubuntu_11-44-42_14-Oct-2016.tar.gz
-rw-r--r--  1 root root 3.5G Oct 14 12:33 redpitaya_ubuntu_11-55-22_14-Oct-2016.img
-rw-r--r--  1 root root 261M Oct 14 12:33 redpitaya_ubuntu_11-55-22_14-Oct-2016.tar.gz
```

That might indicate that you should use http range requests that way if people's connection drops out or if the script restarts it won't re-download the same bytes.

* curl use the -r flag
* wget use the -c flag

jeras, 2016-10-14:
The no space left on device error is expected. See:
https://github.com/RedPitaya/RedPitaya/blob/master/OS/debian/README.md#reducing-image-size
there might be a better method for zeroing drive empty space, but this is the most common.

Where are images being downloaded, there are some instances of curl, but I do not think any of them is loading a whole image.

closed 2023-07-03 by b-swat

andrewhodel, 2023-07-03:
cat was not executed, closing this is invalid.

These are invalid names, it's not right to put the problem on another name because it doesn't have the problem.  When something is killed because it made a big mistake the report should show the truth.

jeras, 2023-07-03:
If this is about SD card creation, there is a better approach than mounting a SD card image and writing to it.
My reply might be unrelated to the issue, I just wrote it here since I was procrastinating too long.

It is possible to create the SD card image without root permissions, thus can be run in CI.
1. `parted` is used to partition the SD card image file,
2. `mkfs.vfat` + `mcopy` are used to create the FAT partition file,
3. `mkfs.ext4` with `-d` argument creates the rootfs partition from a folder, must be done inside `fakeroot` to keep correct permissions,
4. the FAT and rootfs partition files are copied into the SD card image file into the correct positions using `dd`.

With this approach there is no need to zero out the empty part of rootfs partition, since it is cleanly created from a folder.

I got this working for MPSoC+, so this is not just a hypothetical.
I will try to provide some code, but I do not have the environment to test it now.

andrewhodel, 2023-07-03:
I command this is not about SD card creation.

This is about the script `OS/debian/build.sh` creating a new file each time a TCP socket of an HTTP session is blocked or ended by the network.

andrewhodel, 2023-07-03:
This is about allowing an Internet Service Provider the ability to fill the disk.

andrewhodel, 2023-07-03:
The reason this is constructed evil is because the ISP and the hosting server can both be blamed for exploiting the bug and neither maintain record of what would prove those actions.

The bug needs to be fixed.

b-swat, 2023-07-03:
@andrewhodel 
Hi.  I moved this code for creating an OS for RP to another repository.
https://github.com/RedPitaya/ubuntu
If you think that there is something not working correctly, you can create a pull request, I will accept it.

andrewhodel, 2023-07-03:
Moving the code isn't relevant, the code in the latest commit at this date is the same code that you moved.

In reality, you copied it there.  I'm done being persecuted by keyboard for you.

If you look at the commits on and before the date this issue was created, you can easily view the actual file in question and with Github you can use `blame` to see what changes have been made to the file since that date as the issue was not worked on in order with the code changes of this repository.

Here are the commits one month before:

```
commit 1655d53c23d7578aad95b5859e52ba2e45804826
Author: Miha Cankar <miha.cankar@redpitaya.com>
Date:   Sun Oct 16 16:57:22 2016 +0200

    added OS

commit 08ff3e8e16079e305a0a0fd9dc714c490bfbab37
Author: Miha Cankar <miha.cankar@redpitaya.com>
Date:   Sun Oct 16 16:26:18 2016 +0200

    added hardwre

commit 466d06625dda8e84ca29a08438a4cf7ff2811469
Author: Miha Cankar <miha.cankar@redpitaya.com>
Date:   Sat Oct 15 16:45:52 2016 +0200

    qucik start and
    FPGA sections
    done

commit 3b539b511c74fbc61a81ae966a119ef644cd8fcb
Author: Miha Cankar <miha.cankar@redpitaya.com>
Date:   Thu Oct 13 14:36:26 2016 +0200

    TOC done

commit 2d3d1d55ddb13697247c025b8821d4b3a64c196e
Author: amike80 <amike80@users.noreply.github.com>
Date:   Mon Oct 10 08:36:26 2016 +0200

    Initial commit

commit 63382b11b850cfdcd0269ac02101dd0079209e87
Merge: db4dc2954 deec4e80f
Author: Iztok Jeras <iztok.jeras@redpitaya.com>
Date:   Thu Nov 10 12:54:57 2016 +0100

    Merge branch 'doc'

commit deec4e80fe30ce8bf9bc8011540888f42e3960d0
Merge: c301827f6 4d13a3fe2
Author: Iztok Jeras <iztok.jeras@redpitaya.com>
Date:   Thu Nov 10 12:54:07 2016 +0100

    Merge branch 'doc' of ssh://gitlab.redpitaya.com:8022/redpitaya-2-0/redpitaya-public into doc

commit db4dc29548f7c954fdd56fdae684da01f3b5cff3
Merge: 76070756e beb67ee30
Author: Iztok Jeras <iztok.jeras@redpitaya.com>
Date:   Thu Nov 10 11:09:44 2016 +0100

    Merge remote-tracking branch 'origin/release_97_dev'

commit beb67ee30a5f92eda7a6682547b49b145240677b
Author: Mikhail Efimenko <m.efimenko@integrasources.com>
Date:   Wed Nov 9 14:21:28 2016 +0700

    Disable help-system

commit 76070756e4ff80aa935a64edd364bc0dba02c5a1
Author: Iztok Jeras <iztok.jeras@redpitaya.com>
Date:   Tue Nov 8 16:41:47 2016 +0100

    Kernel: for now compiling a shanging branch

commit 464dcfb0f726888c8b6781a208619478e32c5997
Author: Mikhail Efimenko <m.efimenko@integrasources.com>
Date:   Tue Nov 8 14:52:03 2016 +0700

    SCPI acq_trig fix
```

This is a list of the files from the commit in the link - https://github.com/RedPitaya/RedPitaya/tree/63382b11b850cfdcd0269ac02101dd0079209e87

If you view the `OS/debian/` directory it is obvious that this Github repository has been manipulated as the file in the issue is not present.

The reality is that each commit hash should have been posted to Bitcoin as OP_RETURN data in group checksums daily.

Who bought or stole the company and ruined it?