RedPitaya/RedPitaya #81: scpi signal generation

alexanderschuetz, 2016-10-21:
I'm running OS 0.96-86. There was a bug with the PWM generation using SCPI (http://forum.redpitaya.com/viewtopic.php?f=14&t=1044). Will it be fixed soon?

Cheers,
Alex

jeras, 2016-10-21:
sorry, we do not plan to fix this soon

PacoLM, 2019-09-24:
Hello, just acquired a RedPitaya STEM152-10 and need to have the PWM working using SCPI. 
Has this issue been fixed? I have been searching a working example and tried it myself later without success. The signal generation of Sine or Square it works, but I need the PWM.
Regards,
PacoLM

Tetane, 2019-11-14:
I ran into the same issue (PWM not working using SCPI) on a recent version of a Red Pitaya 125-10.
I found an easy workaround which is to send a custom waveform ( https://redpitaya.readthedocs.io/en/latest/appsFeatures/examples/genRF-exm4.html ).

ylectron, 2019-11-14:
Here is ecosystem with PWM bug solved with example how to use it.
https://www.dropbox.com/sh/s55nj9qo8vbeuvp/AACJ64HXCGn4aPSollo4R3fCa?dl=0

b-swat, 2023-07-03:
PWM works in the latest ecosystem.
I used this script

`#!/usr/bin/python3

import sys
import redpitaya_scpi as scpi

rp_s = scpi.scpi('200.0.0.56')

wave_form = 'sine'
freq = 10000
ampl = 1

rp_s.tx_txt('GEN:RST')

rp_s.tx_txt('SOUR1:FUNC PWM')
rp_s.tx_txt('SOUR1:DCYC 0.7')
rp_s.tx_txt('SOUR1:FREQ:FIX 200')
rp_s.tx_txt('SOUR1:VOLT 0.5')
rp_s.tx_txt('SOUR1:VOLT:OFFS 0.5')
rp_s.tx_txt('OUTPUT1:STATE ON')

rp_s.tx_txt('SOUR1:TRIG:INT')
`

closed 2023-07-03 by b-swat