RedPitaya/RedPitaya #85: FPGA: Project creation fails due to missing system.tcl

sanojb, 2016-11-24:
Running "make project PRJ=test" for the current master results in an error

> couldn't read file "ip/system.tcl": no such file or directory

I am new to FPGA programming, but I guess, the script is not being copied properly during project creation. What would be the correct system.tcl to use for a new project?

jeras, 2016-11-24:
there is no project named `test` so option `PRJ=test` will not work
available projects are named the same as subdirectories inside `fpga/prj`

closed 2016-11-24 by jeras

sanojb, 2016-11-24:
Okay, now I seem to slowly get a grasp on the process. The documentation should REALLY be updated with a description of what project to use for what purpose.