RedPitaya/RedPitaya #86: ext4 filesystem on released image not unmounted cleanly

ackalker, 2016-11-26:
After downloading and unzipping [this official image](http://downloads.redpitaya.com/downloads/red_pitaya_OS-v0.96-RC2-76-22_aug.img.zip) from the Red Pitaya site and mounting the OS partition in the image, I noticed that the filesystem wasn't unmounted cleanly: mount wanted to repair it, which failed initially because I had mounted the image file itself read-only:
```sh
$ udisksctl loop-setup -r -f red_pitaya_OS-beta.img
Mapped file red_pitaya_OS-beta.img as /dev/loop0.
$ # can't use udisksctl to mount partition because of https://bugs.freedesktop.org/show_bug.cgi?id=98858
$ sudo mount -o ro /dev/loop0p2 /mnt
mount: wrong fs type, bad option, bad superblock on /dev/loop0p2,
       missing codepage or helper program, or other error

       In some cases useful info is found in syslog - try
       dmesg | tail or so.
```
[from dmesg]
```
[33824.747555] EXT4-fs (loop0p2): INFO: recovery required on readonly filesystem
[33824.747556] EXT4-fs (loop0p2): write access unavailable, cannot proceed
```
I've encountered this problem before with other disk images which were generated by shell scripts: the only workaround I've found which seems to mitigate this somewhat is to add one or more `sync` commands (perhaps interspersed with `sleep 1` commands) before the final `umount`.

jeras, 2016-11-26:
Hi,

This image contains Wyliodrin, which can not be installed with a script on a PC. We have to run wyliodrin.sh directly on the board. So is it possible I did not power down properly after the install process.

Otherwise, one of the last stages of the image creation process is filling the partition with a file containing zeros, so it compresses better. I will check if an extra `sync` would help there.

ackalker, 2016-11-27:
Thanks for the quick reply, I'm looking forward to your findings on this.

jeras, 2016-12-14:
I created a script for checking the image file without writing it onto a SD card.
[OS/debian/image-fsck.sh](https://raw.githubusercontent.com/RedPitaya/RedPitaya/master/OS/debian/image-fsck.sh)

I downloaded the latest stable release and checked it, it seems without errors.
http://downloads.redpitaya.com/downloads/red_pitaya_OS-stable.img.zip
http://downloads.redpitaya.com/downloads/red_pitaya_OS-v0.96-RC1-20-14_jul.img.zip

```
$ sudo OS/debian/image-fsck.sh red_pitaya_OS-v0.96-RC1-20-14_jul.img
fsck.fat 3.0.28 (2015-05-16)
/dev/loop0p1: 616 files, 13072/60289 clusters
e2fsck 1.42.13 (17-May-2015)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
/dev/loop0p2: 43632/216432 files (0.1% non-contiguous), 275244/864768 blocks
```

The beta image does have errors but apparently only on the FAT32 partition, not on ext4. I also noticed it was probably not zeroed properly, and the name was not modified compared to the used base image. We will be more carefull with the next beta.
http://downloads.redpitaya.com/downloads/red_pitaya_OS-beta.img.zip
http://downloads.redpitaya.com/downloads/red_pitaya_OS-v0.96-RC2-76-22_aug.img.zip

```
$ sudo OS/debian/image-fsck.sh red_pitaya_OS-beta.img
fsck.fat 3.0.28 (2015-05-16)
0x25: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
 Automatically removing dirty bit.
Leaving filesystem unchanged.
/dev/loop0p1: 791 files, 13284/60289 clusters
e2fsck 1.42.13 (17-May-2015)
Warning: skipping journal recovery because doing a read-only filesystem check.
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
/dev/loop0p2: 43898/216432 files (0.1% non-contiguous), 290298/864768 blocks
```

closed 2016-12-15 by jeras