RedPitaya/RedPitaya #88: FPGA Firmware

eflyzhao, 2016-12-11:
I tried to load the FPGA firmware using "red_pitaya_vivado_project.tcl", which runs the tcl file at
"x:\RedPitaya\fpga\prj\logic\ip\system.tcl".

Then the errors below appeared.

ERROR: [BD 41-237] Bus Interface property CLK_DOMAIN does not match between /axis_clock_converter_i1/S_AXIS(system_S_AXI_STR_RX1_aclk) and /S_AXI_STR_RX1(system_S_AXI_STR_RX2_aclk)
ERROR: [BD 41-237] Bus Interface property CLK_DOMAIN does not match between /M_AXI_STR_TX1(system_M_AXI_STR_TX1_aclk1) and /axis_clock_converter_o1/M_AXIS(system_M_AXI_STR_TX1_aclk)
ERROR: [BD 41-1031] Hdl Generation failed for the IP Integrator design C:/Users/EflyZhao/Documents/GitHub/RedPitaya/fpga/prj/project/redpitaya.srcs/sources_1/bd/system/system.bd 

Did I load the right "system.tcl"? 
What are the differences among the folders "logic", "logic_origin", "classic", "tft" and "v0.94" in "x:\RedPitaya\fpga\prj"?

jeras, 2016-12-12:
The TCL script `red_pitaya_vivado_project.tcl` is intended for creating a Vivado project for one of our FPGA images, each othe the FPGA image options has a directory in `fpga/prj/`.

The proper way to use it:
```
. setting.sh
cd fpga
make project PRJ=classic
```
You can check the makefile to understand how exactly it is using `red_pitaya_vivado_project.tcl`.

We are still working on the documentation which will describe the functionality provided by each image.

eflyzhao, 2016-12-12:
Thanks for the reply. I am looking for the fpga firmware, which describes the data streams from ADCs and the data streams to DACs. 

So I can do some work on captured ADC data and generate required DAC outputs in FPGA. Could you pliz instruct where the firmware is included? Thank you.

jeras, 2016-12-13:
You should start with the `classic` firmware, which is used by applications like oscilloscope and spectrum analyzer and the API. Data acquisition is limited to 16k samples.

The `logic_orig` firmware is intended for the logic analyzer, it features a DMA to the main mamory, and is supported by API2.

The `logic` firmware is work in progress intended to provide DMA to ADC, DAC and logic analyzer channels.

The `v0.94` firmware is intended for old free unmaintained applications.

xiaoweihua, 2017-05-15:
Hello, jeras:
    I have build the fpga project with vivado2016.4, I work under windows10, and I want to test the hardware but I can not find the bare metal software to start.  I also find that the classic FPGA project have no DMA engine, I really want to try DMA transfer with bare metal software.  can you help me, jeras?

closed 2023-07-03 by b-swat