RedPitaya/RedPitaya #89: problem with direct memory access in v0.97

pavel-demin, 2016-12-17:
Dear Red Pitaya developers,

I've just tested one of the recent SD card images (red_pitaya_OS-v0.97-f9094af-release.zip) and found that the direct memory access in my MCPHA application doesn't work with this SD card image. It works fine with v0.95.

My FPGA configuration is writing data to RAM via S_AXI_HP0 and the C program is reading the data starting from the address 0x1e000000.

In v0.97, some parts of the data are corrupted.

I suspect that the source of the problem is in the PCW_S_AXI_HP0_DATA_WIDTH parameter set to 32 in system.tcl:
https://github.com/RedPitaya/RedPitaya/blob/master/fpga/prj/logic/ip/system.tcl#L991

Similar problems were reported in the following Xilinx forum threads:
https://forums.xilinx.com/t5/Embedded-Linux/Corrupted-data-received-from-AXI-DMA-via-HP0-in-32-bit-mode/td-p/715245
https://forums.xilinx.com/t5/Zynq-All-Programmable-SoC/AXI-DMA-IP-Issues-with-Zynq-Zedboard-Running-Linux/td-p/412683

Could you please set PCW_S_AXI_HP0_DATA_WIDTH to 64?

Best regards,

Pavel

HrRossi, 2016-12-17:
Hi Pavel, 
I'm also looking for the cause of incompatibilities in the AXI HP area with post-0.95 ecosystems, so I'd like to know if you already tried to change the setting and got better results?
Thanks, 
Nils

pavel-demin, 2016-12-17:
Hi Nils,

>  I'd like to know if you already tried to change the setting and got better results?

Not directly. I didn't rebuilt the boot.bin and devicetree.dtb from the fpga/prj/logic source code. I don't have the required toolchain (2016.2, etc).

Copying boot.bin and devicetree.dtb from one of my projects to the v0.97 SD card solved the problem of the memory corruption. Then checked devicetree.dts and system.tcl and the only suspicious code what I found was `PCW_S_AXI_HP0_DATA_WIDTH {32}`. Since similar problems were already reported on the Xilinx forum, I think that it's most probably the source of the memory corruption problem in my application.

It would be great if you could confirm (or not) my suspicions by rebuilding boot.bin and devicetree.dtb from the fpga/prj/logic source code with PCW_S_AXI_HP0_DATA_WIDTH set to 64.

Best regards,

Pavel

pavel-demin, 2016-12-18:
I've tried to build fpga/prj/logic with 2016.3 but I've got some error messages about bus width mismatch and about system.tcl version mismatch.

Then I've installed 2016.2 and tried to build fpga/prj/logic again but I've got the following error messages:
```
ERROR: [BD 41-237] Bus Interface property CLK_DOMAIN does not match between /axis_clock_converter_i1/S_AXIS(system_S_AXI_STR_RX1_aclk) and /S_AXI_STR_RX1(system_S_AXI_STR_RX2_aclk)
ERROR: [BD 41-237] Bus Interface property CLK_DOMAIN does not match between /M_AXI_STR_TX1(system_M_AXI_STR_TX1_aclk1) and /axis_clock_converter_o1/M_AXIS(system_M_AXI_STR_TX1_aclk)
```

I'm using the following commands to build fpga/prj/logic:
```
source /opt/Xilinx/Vivado/2016.2/settings64.sh
source /opt/Xilinx/SDK/2016.2/settings64.sh
git clone git@github.com:RedPitaya/RedPitaya.git
cd RedPitaya/fpga
make PRJ=logic all
```

HrRossi, 2016-12-18:
The "logic" project is not ready for consumption yet, afaik. "logic_orig" is the base of the current builds.

pavel-demin, 2016-12-18:
Nils, thanks for the hint.

I can now build logic_orig. Unfortunately, with PCW_S_AXI_HP0_DATA_WIDTH set to 64, logic_orig doesn't fit into available FPGA resources. This code uses the DMA modules provided by Xilinx and these modules require a lot of FPGA resources.

Since PCW_S_AXI_HP0_DATA_WIDTH is already set to 64 in fpga/prj/classic, I've built boot.bin from this code and copied it to the v0.97 image. With this new boot.bin, the direct memory access in my MCPHA application works without any problem.

I's say that it clearly indicates that the problem is in FSBL. There is something wrong in FSBL when PCW_S_AXI_HP0_DATA_WIDTH is set to 32.

To better understand what happens with FSBL when when PCW_S_AXI_HP0_DATA_WIDTH is set to 32, I've compared FSBL with PCW_S_AXI_HP0_DATA_WIDTH set to 32 and to 64 for a very simple project. Here is the difference:

```
# diff -r led_blinker_64.fsbl/ps7_init.c led_blinker_32.fsbl/ps7_init.c
4153a4154,4163
>     // .. .. n32BitEn = 1
>     // .. .. ==> 0XF8008000[0:0] = 0x00000001U
>     // .. ..     ==> MASK : 0x00000001U    VAL : 0x00000001U
>     // .. ..
>     EMIT_MASKWRITE(0XF8008000, 0x00000001U ,0x00000001U),
>     // .. .. n32BitEn = 1
>     // .. .. ==> 0XF8008014[0:0] = 0x00000001U
>     // .. ..     ==> MASK : 0x00000001U    VAL : 0x00000001U
>     // .. ..
>     EMIT_MASKWRITE(0XF8008014, 0x00000001U ,0x00000001U),
8485a8496,8505
>     // .. .. n32BitEn = 1
>     // .. .. ==> 0XF8008000[0:0] = 0x00000001U
>     // .. ..     ==> MASK : 0x00000001U    VAL : 0x00000001U
>     // .. ..
>     EMIT_MASKWRITE(0XF8008000, 0x00000001U ,0x00000001U),
>     // .. .. n32BitEn = 1
>     // .. .. ==> 0XF8008014[0:0] = 0x00000001U
>     // .. ..     ==> MASK : 0x00000001U    VAL : 0x00000001U
>     // .. ..
>     EMIT_MASKWRITE(0XF8008014, 0x00000001U ,0x00000001U),
12748a12769,12778
>     // .. .. n32BitEn = 1
>     // .. .. ==> 0XF8008000[0:0] = 0x00000001U
>     // .. ..     ==> MASK : 0x00000001U    VAL : 0x00000001U
>     // .. ..
>     EMIT_MASKWRITE(0XF8008000, 0x00000001U ,0x00000001U),
>     // .. .. n32BitEn = 1
>     // .. .. ==> 0XF8008014[0:0] = 0x00000001U
>     // .. ..     ==> MASK : 0x00000001U    VAL : 0x00000001U
>     // .. ..
>     EMIT_MASKWRITE(0XF8008014, 0x00000001U ,0x00000001U),
```

Apparently, these lines are responsible for the problem with the direct memory access.

HrRossi, 2016-12-18:
I'd say it's not exactly wrong, because the logic_orig is indeed using 32bit transfers - it's just incompatible with the 64bit wide transfers most other existing solutions with AXI HP are using. 

However, it is probably possible to reconfigure the AXI HP settings at runtime, as no other components are using the interfaces. Thanks for doing the heavy lifting in this case, I'll also look into it when I'm back to my development environment.

pavel-demin, 2016-12-18:
> I'd say it's not exactly wrong, because the logic_orig is indeed using 32bit transfers - it's just incompatible with the 64bit wide transfers most other existing solutions with AXI HP are using.

I agree that from the ZYNQ configuration point of view it's probably OK to have the HP interface's width set to 32 in FSBL. The problem is that it breaks compatibility with the contributed applications and it's not clear how the n32BitEn bits could be unset from Linux at runtime.

Personally, as an author of some of the contributed applications, I find it quite discouraging that the last two releases (0.96 and 0.97) introduce incompatibilities (https://github.com/RedPitaya/RedPitaya/issues/77 and https://github.com/RedPitaya/RedPitaya/issues/89) with my applications and it takes months for the fixes to be released.

I think that setting PCW_S_AXI_HP0_DATA_WIDTH to 64 for all the applications is the easiest solution. 64 is the default value and it's compatible with all the contributed applications.

To confirm the conclusion from my previous comment, I've built two boot.bin files based on the logic_orig code:
 - boot_32.bin with the unmodified FSBL code
 - boot_64.bin with all the `n32BitEn = 1 ... EMIT_MASKWRITE(...)` lines commented out

The FPGA bitstream (red_pitaya.bit) and the U-Boot executable (u-boot.elf) are exactly the same in both files.

The files can be found at the following link:
https://www.dropbox.com/sh/5fy49wae6xwxa8a/AACp4a_5fuEtWC6P9DU1rrxpa/boot_0_97?dl=0

If I boot Red Pitaya from the v0.97 SD card with the default boot.bin or with boot_32.bin, the direct memory access in my MCPHA application doesn't work properly and some bytes are corrupted.

If I boot Red Pitaya with boot_64.bin, the direct memory access works.

jeras, 2016-12-18:
First what can we do with `logic_orig`. The logic analyzer code is using DMA [2] (AXI HP [2]) and oscilloscope is supposed to use ports 0,1. But since the oscilloscope path was never finished and is therefore not used, modifications there are possible. You already found out why 32 bit buses were used, 64 bit DMA and interconnect consume too much resources.

Second, about modifying the bus width setting at runtime. The PS side of HP ports is configured by FSBL. The question is whether the kernel can reconfigure this ports, and if it can be done using the device tree. We might find the related kernel code by searching for accesses to the relevant register.
In any case I should read some relevant documentation in the TRM.
https://www.xilinx.com/support/documentation/user_guides/ug585-Zynq-7000-TRM.pdf

Third, we are using one of Pavel's applications for our HAMLAB product and we have not encountered this issue yet. Pavel, does the `sdr_transceiver_hpsdr` use the AXI HP 0 port?

HrRossi, 2016-12-18:
The configuration registers in the 0xf8000000 region can be written to like other memorymapped resources, even from userspace (with root priviledges for /dev/mem). 
There is a special unlock / relock procedure to follow to access the SLCR, but that's just more writing to registers. It's all explained in the TRM (chapter B.4 / B.28).

pavel-demin, 2016-12-18:
Hi Iztok,

> Pavel, does the sdr_transceiver_hpsdr use the AXI HP 0 port?

No, sdr_transceiver_hpsdr doesn't use the AXI HP 0 port.

> Second, about modifying the bus width setting at runtime. 

I've tried to check the bits at the addresses 0XF8008000, etc with the following script:
```
for addr in 0XF8008000 0XF8008014 0XF8009000 0XF8009014 0XF800A000 0XF800A014 0XF800B000 0XF800B014
do
  devmem2 $addr
done
```

The bits are set when booting with the default boot.bin or with boot_32.bin (unmodified FSBL):
```
Value at address 0xF8008000 (0xb6fbb000): 0x1
Value at address 0xF8008014 (0xb6f12014): 0xF01
Value at address 0xF8009000 (0xb6f5c000): 0x1
Value at address 0xF8009014 (0xb6fe1014): 0xF01
Value at address 0xF800A000 (0xb6fc9000): 0x1
Value at address 0xF800A014 (0xb6f61014): 0xF01
Value at address 0xF800B000 (0xb6fd3000): 0x1
Value at address 0xF800B014 (0xb6f91014): 0xF01
```

The bits are unset when booting with boot_64.bin (modified FSBL):
```
Value at address 0xF8008000 (0xb6f31000): 0x0
Value at address 0xF8008014 (0xb6efc014): 0xF00
Value at address 0xF8009000 (0xb6f06000): 0x0
Value at address 0xF8009014 (0xb6f0c014): 0xF00
Value at address 0xF800A000 (0xb6fe8000): 0x0
Value at address 0xF800A014 (0xb6f09014): 0xF00
Value at address 0xF800B000 (0xb6f4d000): 0x0
Value at address 0xF800B014 (0xb6fb0014): 0xF00
```

Then I've tried to unset the bits at runtime with the following commands:
```
devmem2 0xF8000008 h 0xDF0D
devmem2 0xF8008000 w 0x0000
devmem2 0xF8008014 w 0x0F00
devmem2 0xf8000004 h 0x767B
```

If I run these commands just after boot, then the HP0 configuration changes and the direct memory access in my MCPHA application works. But setting the bits back to 1 after starting and stopping my MCPHA application doesn't seem to work.

So, looks like the bits can be set only before any activity on the HP bus.

Best regards,

Pavel

HrRossi, 2016-12-18:
> So, looks like the bits can be set only before any activity on the HP bus.
You may need to reset the interface by way of writing 0 to the appropriate bit in 0xf8000240 - this is what the short sentence "The associated "FPGA_RST_CTRL.FPGA_AXDSN_RST" register field
must be written with "0" before accessing this register" in chapter B.4 before each of the register descriptions means.

pavel-demin, 2016-12-18:
> You may need to reset the interface by way of writing 0 to the appropriate bit in 0xf8000240

Also found it and tried. Doesn't seem to help. Here are the commands:

```
devmem2 0xF8000008 h 0xDF0D
devmem2 0xF8000240 w 0x0000
devmem2 0xF8008000 w 0x0001
devmem2 0xF8008014 w 0x0F01
devmem2 0xf8000004 h 0x767B
```

I've also tried several other combinations but with the same result.

HrRossi, 2016-12-18:
Have you tried to query the write protection status (0xf800000c) after writing the magic unlock word? It may be that unlocking wasn't successful. 
Writing to the AXI HP configuration does not require the unlock, but the reset does.

pavel-demin, 2016-12-18:
> Have you tried to query the write protection status (0xf800000c) after writing the magic unlock word?

Checked the status (0xf800000c) after each unlock and lock. The status is always correct.

Strangely, the SLCR module in unlocked by default (after boot).

> I'm not sure using 'h' accesses is correct

I've copied it from XAPP1266 and it seems to work.

After all, I think that I've found what was wrong with my previous tests. If I set n32BitEn to 1 and try to write 64 bits, then the HP bus starts to misbehave.

If I do 32-bit writes when n32BitEn is 1 and 64-bit writes when n32BitEn is 0, then HP bus works correctly and I can switch the modes.

So, it's not foolproof, but it works.

I'll add the following code to my MCPHA application:
```
#include <stdio.h>
#include <stdint.h>
#include <stdlib.h>
#include <unistd.h>
#include <fcntl.h>
#include <sys/mman.h>

int main()
{
  int fd;
  volatile uint32_t *reg;
  volatile uint32_t *axi;

  if((fd = open("/dev/mem", O_RDWR)) < 0)
  {
    perror("open");
    return EXIT_FAILURE;
  }

  reg = mmap(NULL, sysconf(_SC_PAGESIZE), PROT_READ|PROT_WRITE, MAP_SHARED, fd, 0xF8000000);
  axi = mmap(NULL, sysconf(_SC_PAGESIZE), PROT_READ|PROT_WRITE, MAP_SHARED, fd, 0xF8008000);

  reg[2] = 0xDF0D;
  reg[144] = 0;
  axi[0] &= ~1;
  axi[5] &= ~1;
  return 0;
}
```

Since the SLCR module is unlocked after the boot, I'm not locking it in my code.

HrRossi, 2016-12-18:
Great!

> If I do 32-bit writes when n32BitEn is 1 and 64-bit writes when n32BitEn is 0, then HP bus works correctly and I can switch the modes.

It may be prudent then to restore the previous settings when your app unloads / the mcpha process stops.

pavel-demin, 2016-12-19:
> It may be prudent then to restore the previous settings when your app unloads / the mcpha process stops.

I agree but I don't see any easy and robust solution.

I'd say that each application should configure the axi_hp registers before it starts transfers.

HrRossi, 2016-12-19:
Or the nginx-plugin could be adapted to read a configuration value from the fpga.conf to make the required setting before loading an app.

pavel-demin, 2016-12-19:
> Or the nginx-plugin could be adapted to read a configuration value from the fpga.conf to make the required setting before loading an app.

It would be ideal.

jeras, 2016-12-19:
Nginx is only one way to start an application, so solving it there would not be the best solution.

I see using device tree overlays as the best long term solution. They are intended for runtime reconfigured hardware. Our current kernel already has patches for overlays, and loading a device tree overlay works properly (loading FPGA bitstreams through overlays was never tested properly, but might work with few changes). But I doubt the code for reconfiguring AXI HP ports is present in the kernel. I spent about 1 hour searching for related code in the kernel but could not find it.

Another obvious solution would be to modify our FPGA images to use 64bit AXI HP ports, but this will take some time, since I have to somehow reduce code size while preferably avoiding SW changes to minimize testing. Zynq TRM clearly states 32bit HP bus mode has other performance restrictions, not only the bus width.

HrRossi, 2016-12-19:
> Nginx is only one way to start an application, so solving it there would not be the best solution.

But it would cover a significant swath of usecases. Applications that are started from the command line will need to take action themselves, of course. 

> I see using device tree overlays as the best long term solution.

The devicetree is informing the kernel about the layout of the existing devices. I wouldn't see reconfiguration of a specific piece of hardware as a task for a dtbo. 
And applications would still need to actively load their required dtbo, so it's not really an advantage over doing the configuration by some other means.

HrRossi, 2016-12-20:
OK, after taking a closer look at the different ZYNQ drivers it is clear that Xilinx likes to have configuration presets in the devicetree. I'd say it would make sense to follow their lead. 

The AXI HP interfaces don't have dedicated drivers, I guess because they are just internal connections and they don't fit the categories platform, bus, device. 
The dma drivers already support a "xlnx,datawidth" dt-property, but I don't see how the dma drivers could find out to which AXI port they are connected in order to use the property to reconfigure it. Of course, we could add another dt-property to them, then patch the devicetree-generating tcl to extract the information from the block design and put it there. 
In my opinion, it would be easier to do it programmatically. Put this into nginx and the api's rp_Init() and add a note to the docs that applications that don't use either need to reconfigure the interfaces first if they want to use them. 

ps @pavel-demin it seems that the "xlnx,zynq-slcr" driver unlocks the SLCR early during boot and leaves it unlocked.

closed 2023-07-03 by b-swat