RedPitaya/RedPitaya #98: Cannot build EcoSystem - make: *** No rule to make target 'fpga.zip', needed by 'build/fpga'

XYEFA, 2017-01-23:
Hello,

I am trying to build the RedPitaya EcoSystem and it is failing on a build rule. Is the build tree broken? I am building on an Ubuntu 16.04 Machine. Vivado is 2016.4

make -f Makefile.x86 install

...
...

```
cp tmp/linux-xlnx-redpitaya-v2016.2/arch/arm/boot/uImage build
curl -L https://github.com/Xilinx/device-tree-xlnx/archive/xilinx-v2016.4.tar.gz -o dl/device-tree-xlnx-xilinx-v2016.4.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   139    0   139    0     0    181      0 --:--:-- --:--:-- --:--:--   181
100 73750    0 73750    0     0  45436      0 --:--:--  0:00:01 --:--:--  385k
mkdir -p tmp/device-tree-xlnx-xilinx-v2016.4
tar -zxf dl/device-tree-xlnx-xilinx-v2016.4.tar.gz --strip-components=1 --directory=tmp/device-tree-xlnx-xilinx-v2016.4
make: *** No rule to make target 'fpga.zip', needed by 'build/fpga'.  Stop.

```

HrRossi, 2017-01-24:
I found out yesterday that it succeeds if you do "make -f Makefile.x86 fpga_all" before, which creates the fpga.zip . Still, it would be nice if the build works out-of-the-box.

XYEFA, 2017-01-24:
I made it a little further and then hit these messages:

```
INFO: [Timing 38-2] Deriving generated clocks [/media/user/ExternalDisk/RedPitaya/fpga/sdc/red_pitaya.xdc:212]
/media/user/ExternalDisk/Vivado/2016.4/bin/loader: line 179: 30929 Killed                  "$RDI_PROG" "$@"
Makefile:38: recipe for target 'prj/classic/out/red_pitaya.bit' failed
make[1]: *** [prj/classic/out/red_pitaya.bit] Error 137
make[1]: Leaving directory '/media/user/ExternalDisk/RedPitaya/fpga'
cp: cannot stat 'fpga/prj/classic/out/red_pitaya.bit': No such file or directory
cp: cannot stat 'fpga/prj/classic/sdk/fsbl/executable.elf': No such file or directory
cp: cannot stat 'fpga/prj/classic/sdk/dts': No such file or directory
cp: cannot create regular file 'build/fpga/classic/dts/': Not a directory
make[1]: Entering directory '/media/user/ExternalDisk/RedPitaya/fpga'
vivado -nolog -nojournal -mode batch -source red_pitaya_vivado.tcl -tclargs logic_orig

---------------------------------------------------------------------------------------------------
INFO: [Synth 8-3354] encoded FSM with state register 'sm_scc_state_reg' using encoding 'sequential' in module 'axi_datamover_scc'
ERROR: [Common 17-217] Failed to load feature 'core'.
ERROR: [Common 17-217] Failed to load feature 'core'.
Parallel Worker Creation Failure: Worker 2 has died.  Skipping 
Parallel Worker Creation Failure: Worker 3 has died.  Skipping 
```

<

jeras, 2017-01-24:
this seems a Vivado crash, it might be an issue with the base OS, a
corrupted Vivado installation or just a random crash (out of memory, ...)

XYEFA, 2017-01-24:
Thanks for the comment - I am building it inside a Virtual Box VM. The performance has been a bit slow. When I do it on a bare metal system it could improve.

Asad-1, 2019-06-28:
@XYEFA @jeras Was there any progress on this? I'm building inside a virtual box VM as well, and keep getting a similar error message. Did it help to switch to a bare metal system?

closed 2023-07-03 by b-swat