RedPitaya/RedPitaya #99: What is rp_CalibInit(); for what and when should I use it?

benalcazardiego, 2017-02-12:
Hello. In rp.h I see this function here:
/**
 * Initializes the library. It must be called first, before any other library method.
 * @return If the function is successful, the return value is RP_OK.
 * If the function is unsuccessful, the return value is any of RP_E* values that indicate an error.
 */
int rp_Init();

int rp_CalibInit();

However it does not have a description. When should I use it?

Thanks

jeras, 2017-02-13:
Hi the API code is a mess of wrapper functions, especially calibration, so I can not tell you for sure by looking at the code. In principle it reads calibration parameters from the EEPROM.

What I can tell you is you do not have to use it, if your aim is to use the generator or acquisition.
I am able to use the API without using this function.

I plan to release a newer version of this api in a couple of days, there will be few incompatible changes, but the code should be much shorter and easier to read. Please expect some calibration issues at first, but acquire and generator should work.

HrRossi, 2017-02-14:
@jeras please consider a different namespace for the new incompatible api, like "sl_" (STEMlab). 
I'm still getting questions why pre-0.94 binaries can't be used. I don't contest a redesign, but there is no need to break compatibility with it.

J65mesW, 2017-02-14:
Hello, is there any change that calibration will be changed? I have some issues with it.

The calibration of outputs is OK, but inputs are little bit difficult. When I calibrate inputs in ScopePro it not work in old scope. I think that one issue is different sign for offsets. For old scope doesn't matter.

The second problem is that calibration depend on filter set to scope module. The users can change filter and it also depend on decimation factor. Another thing is that sometimes is calibration useful for PID, then again it is difficult to use calibration because PID doesn't use filtering (this is OK).

So there should be some clear description what is actually calibrated.

Moreover offsets can be non integer types also. The storage format of gain coefficients is weird for me.
When I want gain I need compute this
g = 100.0*gain_fs/(1<<(32+c_osc_fpga_adc_bits-1))

Fixed point is OK, but why 100, why not 128 or 64? Am I missing something?

jeras, 2017-02-15:
The calibration code is full of changes accumulated over time, and since the values are stored in EEPROM on devices already sold, it is difficult to redo it cleanly.

While it is true the offset could be a fraction of an integer, the error is within quantization noise levels.

I have no idea where the gain encoding comes from, this is from before I joined the company.

The effects of the input filter on calibration can only be fixed by moving the calibration into the FPGA, but this can not be done without API changes. But with our current API it is difficult to change anything without breaking existing applications.

I committed a cleaner version of the API (librp1.so and redpitaya/rp1.h) into the api1/ directory. The code is not good yet, but it is much easier to read and modify. I am using it for Jupyter examples:
https://github.com/RedPitaya/RedPitaya/blob/master/jupyter/examples/signal_acq_stream.ipynb

An image with Jupyter support is available here:
http://downloads.redpitaya.com/ubuntu/redpitaya_ubuntu_13-21-01_14-feb-2017.img.zip

benalcazardiego, 2017-02-17:
Ok. I see that it shall not be considered in my work since I only need to focus on adquisition.

closed 2017-02-22 by benalcazardiego