RedPitaya/RedPitaya #103: Doubt about buffer

benalcazardiego, 2017-02-22:
Hello. I have one simple question. 
The ADC buffer fills continuously. After a trigger is found a counter with 8192+trigger_delay samples is started. When it gets to zero, the source changes to idle. Then I want to get the data with rp_AcqGetOldestDataV(RP_CH_2, &buff_size, buff) to a memory 'buff' with size 'buff_size'=8192+trigger_delay I previously created. So my question is what happens if the trigger is executed when, for example, three quarters of the buffer have been filled? Then the counter cannot get the 8192+delay samples from the ADC  buffer. Does it wait for the last quarter and the first quarter+delay samples? How do I use rp_AcqGetOldestDataV in this case, would it get the the last quarter and the first quarter+delay samples or would it get only the first quarter? How would I avoid losing samples?
Thanks and greetings

HrRossi, 2017-02-26:
If three quarters of the buffer have been filled, the remaining quarter will contain old data (whatever was in the buffer from a previous acquisition). The GetData... functions do not wait nor do they 'know' if the contents of the buffer are new or old. 
This is the main reason for the sleep commands after arming the scope - to allow the buffer to be filled with fresh samples before activating the trigger.

benalcazardiego, 2017-03-03:
I may not expressed well. My doubt is that if the trigger is located at the moment when three quarters of the buffers are filled the counter will start then and continue until about one quarter is filled with new data (Since the counter has 8192 + delay samples). If I use GetOldestdata which data will I be acquiring? Those from the start of the counter?

benalcazardiego, 2017-03-03:
In practice what I need is just to pass all the samples to the ADC buffer and then to the buffer I created. I cannot lose any data. Therefore I set the trigger really low

benalcazardiego, 2017-03-03:
I have a program that works perfectly excepts that I am losing pulses. I know that since I see the pulses that arrive in an oscilloscope simultaneousy. The width of the pulses is about 300 ns (40 samples with decimation 1). Ignore the HV sections. It only is used to turn the sensor on

    
    #include <stdio.h>
    #include <stdlib.h>
    #include <unistd.h>
    #include <time.h> 			
    #include "redpitaya/rp.h"
    #include "math.h"
    #include <inttypes.h>

    #define VERSION "0.1"

    int main (int argc, char **argv) {
	
	/* Syntax analysis */
	if (argc != 5)
	{
		fprintf(stderr,"\tError: Incorrect syntax.\n Syntax:  LD_LIBRARY_PATH=/opt/redpitaya/lib ./%s [Filename] [Trigger Level] [Measure time] [HV 0] \nExample: LD_LIBRARY_PATH=/opt/redpitaya/lib ./%s data.dat -0.2 10 1.2 \n",argv[0],argv[0]);
		return EXIT_FAILURE;
	}
	
	if (atof(argv[4]) < 0 || atof(argv[4]) > 2.5)
	{
		fprintf(stderr,"\tError: Incorrect input value.\n HV must satisfy the condition 0 [V] <= HV <= 2.5 [V]");
		return EXIT_FAILURE;
	}
	
	/* Definition of the input parameters */
	float trigger_level = atof(argv[2]);
	float measure_time = atof(argv[3]);
	float HV = atof(argv[4]);
	
	
	
	/* General information */
	fprintf(stdout,"\t%s version %s\n",argv[0],VERSION);
	fprintf(stdout,"\tThe HV value sent is %f (V)\n",1.4*HV);
	   
    /* Open file and test for error */
    FILE *fd;
    fd = fopen(argv[1], "wb");
    if (fd == NULL) 
	{ 
    	fprintf(stderr, "\tError: Unable to input file %s \n", argv[1]);
    	return EXIT_FAILURE;
	}
    
    /* Create buffer */
    uint32_t buff_size = 8232; /*8192 + 40*/
    float *buff = (float *)malloc(buff_size * sizeof(float));
    
    /* Initialization of the RP */
    if(rp_Init() != RP_OK)
	{
        fprintf(stderr, "\tError: Rp api init failed!\n");
        return EXIT_FAILURE;
    }
    
	/* HV setting */
	rp_AOpinReset();
	float step = 0.001;
	float base = 0;
	float tiempo = 5;
	
	/* Definition of the base value (may be ommited) */
	if(rp_AOpinGetValue(0, &base) != RP_OK)
	{
        fprintf(stderr, "\tError: HV base value establishment failed\n");
        return EXIT_FAILURE;
[... 211 lines cut ...]
    
	fprintf(stdout, "\tEstablished final HV value: %f \n", HV_final_value);
	
    /* Close of the file */
    fclose(fd);
    
    /* Reset of the RP
    if(rp_Reset() != RP_OK)
	{
        fprintf(stderr, "Error: Rp api res failed!\n");
        return EXIT_FAILURE;
    }
    */
    
	/* Release of the RP */
    if(rp_Release() != RP_OK)
	{
        fprintf(stderr, "\tError: Rp api release failed!\n");
        return EXIT_FAILURE;
    }
    
    free(buff);
    fprintf(stdout, "\tThe application has finished working properly\n");
    return EXIT_SUCCESS;
    }

closed 2023-07-03 by b-swat