RedPitaya/RedPitaya #105: Errors Building Red Pitaya OS v0.96-RC2

XYEFA, 2017-02-28:
My build is failing. I previously had success with this build:

```
  HOSTCC  scripts/basic/bin2c
  WRAP    arch/arm/include/generated/asm/unaligned.h
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
  Generating include/generated/mach-types.h
  HOSTCC  scripts/genksyms/genksyms.o
  HOSTCC  scripts/dtc/dtc.o
  CC      scripts/mod/empty.o
gcc: error: unrecognized argument in option ‘-mabi=aapcs-linux’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
gcc: error: unrecognized command line option ‘-mlittle-endian’
gcc: error: unrecognized command line option ‘-mno-thumb-interwork’
gcc: error: unrecognized command line option ‘-mfpu=vfp’
scripts/Makefile.build:258: recipe for target 'scripts/mod/empty.o' failed
make[3]: *** [scripts/mod/empty.o] Error 1
scripts/Makefile.build:403: recipe for target 'scripts/mod' failed
make[2]: *** [scripts/mod] Error 2
make[2]: *** Waiting for unfinished jobs....
  HOSTCC  scripts/dtc/flattree.o
  CC      kernel/bounds.s
gcc: error: unrecognized argument in option ‘-mabi=aapcs-linux’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
gcc: error: unrecognized command line option ‘-mlittle-endian’
gcc: error: unrecognized command line option ‘-mno-thumb-interwork’
gcc: error: unrecognized command line option ‘-mfpu=vfp’
Kbuild:44: recipe for target 'kernel/bounds.s' failed
make[2]: *** [kernel/bounds.s] Error 1
Makefile:986: recipe for target 'prepare0' failed
make[1]: *** [prepare0] Error 2
make[1]: *** Waiting for unfinished jobs....
```

jeras, 2017-02-28:
it seems you are compiling with x86 compiler
before you start the compile process, source settings file:
$ . settings.sh
there is 0.97-RC4 tag, which was released today,
if you do not need the specific old build I recommend the new one

XYEFA, 2017-02-28:
Yes - Thanks - That seems good.

XYEFA, 2017-03-02:
I have another issue while I am on this topic. When I try to build the Ubuntu build I get this error message:

`chroot: failed to run command ‘/bin/bash’: No such file or directory`

It does not seem to stop anything, but how can I fix this?

jeras, 2017-03-02:
This one is too generic, I would need to see the steps you took and some more output.

XYEFA, 2017-03-02:
I saw the following message during the Ubuntu build when I ran the sudo OS/debian/image.sh build script

```
RedPitaya$ sudo OS/debian/image.sh
3500+0 records in
3500+0 records out
3670016000 bytes (3.7 GB) copied, 65.1305 s, 56.3 MB/s
mkfs.fat 3.0.28 (2015-05-16)
unable to get drive geometry, using default 255/63
/dev/loop2p1 has 255 heads and 63 sectors per track,
hidden sectors 0x0000;
logical sector size is 512,
using 0xf8 media descriptor, with 241664 sectors;
drive number 0x80;
filesystem has 2 16-bit FATs and 4 sectors per cluster.

‘patches/fw_env.config’ -> ‘root/etc/fw_env.config’
install: creating directory ‘root/etc/apt’
install: creating directory ‘root/etc/apt/apt.conf.d’
‘OS/debian/overlay/etc/apt/apt.conf.d/99norecommends’ -> ‘root/etc/apt/apt.conf.d/99norecommends’
‘OS/debian/overlay/etc/apt/sources.list’ -> ‘root/etc/apt/sources.list’
‘OS/debian/overlay/etc/fstab’ -> ‘root/etc/fstab’
‘OS/debian/overlay/etc/hostname’ -> ‘root/etc/hostname’
‘OS/debian/overlay/etc/timezone’ -> ‘root/etc/timezone’
‘OS/debian/overlay/etc/securetty’ -> ‘root/etc/securetty’
**chroot: failed to run command ‘/bin/bash’: No such file or directory**
‘OS/debian/overlay/etc/iptables/iptables.rules’ -> ‘root/etc/iptables/iptables.rules’
install: invalid option -- 'V'
Try 'install --help' for more information.
install: creating directory ‘root/etc/systemd’
install: creating directory ‘root/etc/systemd/network’

```

closed 2023-07-03 by b-swat