RedPitaya/RedPitaya #116: Oscilloscope CSV Export Bugs

sefffal, 2017-06-13:
Hello,
I've found a few bugs in the CSV export in the Oscilloscope & Signal Generator web application.

First, the time units in the resulting CSV are inconsistent between rows. When the absolute magnitude of the time is larger than 1 μs, the value in that row will be in units of  μs. When the absolute magnitude of the time value is less than 1 μs, the value in that row will be in units of ns. If you try and plot the data in the CSV, you will therefore see a very odd plot. As a related issue, the unit reported in the column header appears to be taken from the first row only.

Secondly, the value columns (IN1, IN2) seem to be in units of divisions on the oscilloscope graph, rather than V or mV. Unless the user remembers to write down their zoom settings when they exported the CSV, the data becomes pretty meaningless.

Finally, if you increase the horizontal zoom below 1 μs/divsion, then the resulting CSV will contain interpolated points from the graph instead of just the raw data (e.g. points spaced by 1ns when the limit of the hardware is 8ns). This can make fitting a function to the data quite problematic.

This was tested in software version 0.97-0.97-396.

I can provide examples of CSV files with these issues if that would be helpful.
Thank you,
-Will Thompson

```
##############################################################################
# Red Pitaya GNU/Linux Ecosystem
# Version: 0.97
# Build: 396
# Branch: 
# Commit: 9d1b5c6baa55ffebed6c90df5f8ef295b8c7d76e
# U-Boot: "redpitaya-v2016.4"
# Linux Kernel: "redpitaya-v2016.2-RP0"
# Pro Applications:  3dd4964d8f0ff8cbf2b07335061ea35b849f73da Applications (v0.96-RC3-141-g3dd4964)
##############################################################################
```

sefffal, 2017-06-15:
If someone can point me to the code that does the CSV export, I can look into fixing it.

sefffal, 2017-08-16:
Any updates on this bug? Has anyone looked into it?

mbr99, 2017-11-09:
... i also noticed that...

sbolesjgu, 2018-08-10:
Same bug with my Pitaya, had to fix it with a mathematica code. 

Another thing i get when using the webbrowser oscilloscope is that my time data is reversed...
When I look with the oscilloscope at a "rising signal" and then export my data, i will get the same data but time inversed so I get a "falling signal". 
Fixed that with Mathematica also but right now i'm pretty frustrated with the device because i need to do 2 hours of work to get any usable data...

Does someone also have experienced this bug? Or am i just missing something in the settings