RedPitaya/RedPitaya #118: how to run    apps-free   manully

linozhao, 2017-06-21:
hi,
   how to run    apps-free   manully?
thank you.

linozhao, 2017-06-22:
how can I get     gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf.tar.xz  ???????????

jeras, 2017-06-22:
There is no need to download gcc-linaro, since an appropriate cross compiler is available in Vivado.
It is made available by running:
`$ source settings.sh`

jeras, 2017-06-22:
Regarding building free applications, we do not maintain them any more, and they are not compiled by default. In any case user space applications are not cross-compiled, instead they are compiled in a virtual ARM environment, which means it is almost the same as compiling directly on Red Pitaya.
http://redpitaya.readthedocs.io/en/doctest/doc/developer.html

This should build free applications and create zip files, but as I described, it was not tested recently.
# make apps-free

sefffal, 2017-06-26:
Hi @jeras, could you elaborate on not supporting the free applications anymore? Are you referring to the applications that come with the Red Pitaya?

jeras, 2017-06-26:
I am referring to applications contained in the `apps-free` folder. This applications were written without a common API, meaning, many changes would be required for them to support an updated FPGA image. They also use some other obsoleted components, making them difficult to maintain.

Applications currently available in the release image are free, but not open source (there is usually a `pro` in the app name).

linozhao, 2017-07-09:
could you help me ?
I just want to change some procedure to get my new application.but ,i do not run apps-free manully.

linozhao, 2017-07-11:
when get apps-free from bazaar ,it can run successly.but,howmannuly

closed 2023-07-06 by b-swat