RedPitaya/RedPitaya #119: Web Application - LED Example

anpattividana, 2017-06-26:
This is regarding with LED example of the Web Application section. The example application is not working (Lights are not turn on) until one of other application is started or SCPI server started. After any change of relative area, It will work successfully until reboot. #
What is the reason for this problem ? Missed anything ?

sefffal, 2017-06-26:
I'm not associated with Red Pitaya but I may be able to help.
Any application that interacts with the hardware needs to specify what FPGA image to load when the application is started.
For most purposes, you can create a file called "fpga.conf" in your app directory with the contents "/opt/redpitaya/fpga/fpga_0.94.bit", without the quotes.
I was under the impression that fpga_0.94.bit was loaded upon startup but your issue would seem to indicate not.

jeras, 2017-06-26:
The reason is the FPGA bitstream currently loaded.
The next two bitstreams have a GPIO/LED register in the housekeeping module:
`/opt/redpitaya/fpga/fpga_0.94.bit`
`/opt/redpitaya/fpga/fpga_0.94.bit` (also used by SCPI)
While `classic`, 'logic`, `mercury` images connect the GPIO controller inside PS to drive GPIO/LED.

I am not sure, since it is not my code, but apparently the LED example uses the dedicated GPIO register.

A fpga.conf should be added to the application, so that it will load the correct FPGA code.
`https://github.com/RedPitaya/RedPitaya/blob/master/apps-tools/scpi_manager/fpga.conf`

If you have python experience, you might try Jupyter. It is far easier to make a simple user interface in Jupyter then writing a custom web application.

jeras, 2017-06-26:
Currently `logic` bitstream loads on power up, this is due to some issues bringing up the DMA driver, we hope to fix this soon on master.
In the future, the FPGA might be empty and only loaded when an application is started.

closed 2023-07-06 by b-swat