RedPitaya/RedPitaya #124: Starting the SCPI server while the oscilloscope app is running crashes the STEMLAB

sefffal, 2017-08-16:
If you have the oscilloscope app running in one browser/tab and open the scpi_manager page open in the other, starting the SCPI server crashes the STEMLAB.
This is curious because the SCPI server can be used while the oscilloscope is running, so long as it is started beforehand.

If this is not supported, the button should be grayed out.  It shouldn't be this easy to crash the STEMLAB.

jeras, 2017-08-16:
With the current API, it is not possible to run more then one application simultaneously, at least not reliably, since all applications can access the same HW registers without consideration for other applications.

The crash is actually caused by reloading the FPGA image, which is done while starting the SCPI server.
Applications like oscilloscope access FPGA registers (read data buffer) all the time, and if the FPGA is offline for half a second, such an access causes a bus error. On the other hand while the SCPI server is running it does not mean, any register read/write operations are being done unless SCPI commands are being processed, so starting oscilloscope while SCPI server is up, does not always cause a bus error.

A newer API is beeing worked on, where there is some detection provided regarding what FPGA image is currently loaded. While it is still not possible for more then one app to use the same hardware module, it will be possible to split HW resources among applications. So you could run a web interface oscilloscope and a Python based generator simultaneously. The problem still remains if the two applications do not use the same FPGA bitstream. The last issue can not be solved (maybe with partial reconfiguration, but with a lot of effort and many hard to explain limitations), but it could be managed, by telling the user there is a conflict, before loading a new FPGA image.

sefffal, 2017-08-28:
Hi @jeras, thanks for the info. That is an exciting development.
Will the new API remain backwards compatible with applications developed around the current API?

martsalz, 2018-06-21:
When will the new API be released?

closed 2023-07-06 by b-swat