RedPitaya/RedPitaya #126: Writing to a GPIO?

holachek, 2017-09-05:
I used to use the monitor binary to write to GPIO pins (specifically, DIO0_P):
    monitor 0x40000010 0x01 // set DIO0_P as output
    monitor 0x40000018 0x01 // set DIO0_P HIGH

However, these commands no longer work on the latest Red Pitaya OS. How can I access this GPIO pin another way? I've read through the docs, which I don't understand.

urosgolobrp, 2017-09-06:
Sometime in a summer release last year ( >=0.95 )we adopted Linux kernel gpio subsystem, so on current releases you can use now obsolete sysfs interface. Documentation is available at https://github.com/RedPitaya/RedPitaya/blob/hamlab/fpga/fpga.rst#gpio-and-leds  
For gpios on extension connector E1 you will have to export each pin from [975:968] or [983:976]

`$ export INDEX=975`
`$ echo $INDEX > /sys/class/gpio/export`
`$ echo out    > /sys/class/gpio/gpio$INDEX/direction`
`$ echo 1      > /sys/class/gpio/gpio$INDEX/value`
`$ cat           /sys/class/gpio/gpio$INDEX/value`

You can also find detailed explanation of subsystem on zynq at http://www.wiki.xilinx.com/Linux+GPIO+Driver 

But take note that in one of next releases that will come with newer kernel based on 4.10 will come with new gpio subsystem since kernels based on >4.8 use newer gpio subsystem based on character device.

holachek, 2017-09-06:
Thanks for your reply. I tried using the example sequence to set turn on an LED connected to DIO0_N, however am running into problems:
`$ export DN=976
$ echo $DN > /sys/class/gpio/export
$ echo out > /sys/class/gpio/gpio$DN/direction
$ echo 1 > /sys/class/gpio/gpio$DN/value
$ cat /sys/class/gpio/gpio$DN/value
0`

It looks like the system is not actually accepting the value I send to the GPIO.

urosgolobrp, 2017-10-12:
Totally forgot to mention that requirement for using Linux gpio api is that you use appropriate fpga bit-stream. Gpio can not work when using logic analyser bitstream (default at boot) since pins are used by logic analyser software. 
But you can use Linux GPIO with mercury and classic bit streams. You can apply this using
 `cat /opt/redpitaya/fpga/classic/fpga.bit > /dev/xdevcfg` 
Simple blink C example for GPIO is available at: https://github.com/RedPitaya/RedPitaya/tree/master/Examples/gpio_sysfs

closed 2017-10-12 by urosgolobrp