RedPitaya/RedPitaya #132: Jupyter Python script "Blink LED" only works as root user

hubgitti, 2017-12-12:
Copying your jupyter example "Blink LED" to the console does not work with any user but root. Every non-root user gets the error message:

PermissionError: [Errno 13] Permission denied: '/sys/class/leds/led1/brightness'

jeras, 2017-12-12:
While is is possible to write an UDEV rule to get write access rights to LED, they are currently commented out, since they do not work well.
The problem is LED directories are created on demand, and UDEV is not fast enough to change access rights before your software attempts the access, wait time or something similar would be needed.

https://github.com/RedPitaya/RedPitaya/blob/affc65e62f13c0584818953e323edcc6f063982b/doc/developerGuide/os/udev.rst

I am not sure which users are given access rights to hardware, check here:
https://github.com/RedPitaya/RedPitaya/blob/master/OS/debian/zynq.sh#L106

jeras, 2017-12-12:
I do not work for Red Pitaya any more, so do not ask me to fix anything please.

hubgitti, 2017-12-13:
Thanks for your information! I'll try to cope with the issues, and I'm curious how things will develop here...

b-swat, 2023-07-06:
@hubgitti  Is the problem still relevant? ))

closed 2023-07-07 by b-swat