RedPitaya/RedPitaya #135: Jupyter python script "osc_trigger_forced" fails after using the nginx web oscilloscope

hubgitti, 2017-12-12:
After using your nginx web oscilloscope, your nginx web jupyter python interface fails. The jupyter python interface with program "osc_trigger_forced" only displays zero (0.00000) values on the fast input channels. Only a reboot of the red pitaya restores the situation to a working jupyter interface.

hubgitti, 2017-12-13:
Thanks, the need of a reboot "hurts" because it makes it more inefficient to use use your comfortable web oscilloscope. I would like to prepare an experiment or setup with your web solution, and after everything looks fine, I would like to switch to a much less sophisticated self made solution to carry out the actual measurement.

jeras, 2017-12-19:
It is definitely not possible to run more then one application on the same hardware. The mercury FPGA project provides access to each HW module (2 x scope, 2 x generator, logic analyser, ...) using UIO. UIO creates a device file, which can be locked by an application.
So it is possible to lock access to parts of the FPGA by SW. But if you wish to use the same HW using another app, you have to first close the first app, or implement a release mechanism inside it.
In any case this locking mechanism is not implemented in web apps (it is implemented in Python apps, but not tested much).

closed 2023-07-06 by b-swat