RedPitaya/RedPitaya #136: Jupyter scprit "osc_trigger_forced": Calibration in eeprom is not used

hubgitti, 2017-12-12:
It looks to me as if the calibration in the eeprom, generated by your nginx web oscilloscope is not used by your python jupyter script "osc_trigger_forced".

jeras, 2017-12-12:
You are correct. Calibration is not fully implemented in Python yet. While with web applications calibration is done in software, the mercury bitstream used in Jupyter performs calibration in hardware, but for now related registers are only loaded with 0 for offset and 1 for gain. There is a calibration demo in the Jupyter repo that might help you, but note the way calibration values are stored in the EEPROM is super convoluted, you might need a lot of change for the swear jar if you wish to read the related code.

hubgitti, 2017-12-13:
Thanks for the information!

Same here:
To the members still working for RedPitaya: To my humble opinion, this issue needs a fix or solid workaround. Calibration would be quite important to me as I'm using the 14bit version of your StemLab. I "implemented" a simple self made solution in python, but it sounds more sophisticated to me to have that job done by the RedPitaya stuff. I would trust your solution much more than my own workaround.

b-swat, 2023-07-06:
Calibration has been implemented

closed 2023-07-06 by b-swat