RedPitaya/RedPitaya #139: Error armada after exiting python in jupyter example osc_trigger_forced

hubgitti, 2018-01-08:
Using your python example osc_trigger_forced on the console, after exiting python, the following errors are shown on the console:

Overlay __del__ was activated.
But since there are garbage collection ordering issues, this code is commented out.
UIO __del__ was activated.
Exception ignored in: <bound method osc.__del__ of <redpitaya.overlay.mercury.mercury.osc object at 0xb69ce450>>
Traceback (most recent call last):
  File "/home/jupyter/RedPitaya/redpitaya/drv/osc.py", line 62, in __del__
  File "/home/jupyter/RedPitaya/redpitaya/drv/uio.py", line 90, in __del__
BufferError: cannot close exported pointers exist

The errors may show up multiple times, messing with the scripts intentional message output.

hartelt, 2019-05-28:
Same here... any news on this?

chGoodchild, 2020-06-15:
I had the same problem. 

I get this problem the second time that I call the following piece of code
```
from redpitaya.overlay.mercury import mercury as overlay
fpga = overlay()
gen0 = fpga.gen(0)
# Some code
gen0.enable = True
# Some code
gen0.enable = False
```

The exception no longer occurs after removing `gen0.enable = False`

I noticed that this link `/dev/uio/gen0` is present when the error occurs, but deleting the link and/or it's corresponding file in `/dev/uio/.` doesn't resolve the issue.

Restarting the red pitaya does resolve the issue though.