RedPitaya/RedPitaya #151: can't generate bitstream

TinaHsing, 2018-07-03:
Sysstem: Ubuntu
Vivado: 2017.4 Webversion
after excute make PRJ = calssic
the error code is
ERROR: [Synth 8-439] module 'system' not found [/home/tina/Documents/RedPitaya/fpga/prj/mercury/rtl/red_pitaya_ps.sv:97]
WARNING: [Synth 8-693] zero replication count - replication ignored [/home/tina/Documents/RedPitaya/fpga/prj/mercury/rtl/red_pitaya_ps.sv:181]
WARNING: [Synth 8-693] zero replication count - replication ignored [/home/tina/Documents/RedPitaya/fpga/prj/mercury/rtl/red_pitaya_ps.sv:181]
WARNING: [Synth 8-693] zero replication count - replication ignored [/home/tina/Documents/RedPitaya/fpga/prj/mercury/rtl/red_pitaya_ps.sv:181]
ERROR: [Synth 8-285] failed synthesizing module 'red_pitaya_ps' [/home/tina/Documents/RedPitaya/fpga/prj/mercury/rtl/red_pitaya_ps.sv:28]
ERROR: [Synth 8-285] failed synthesizing module 'red_pitaya_top' [/home/tina/Documents/RedPitaya/fpga/prj/mercury/rtl/red_pitaya_top.sv:8]
---------------------------------------------------------------------------------
Finished RTL Elaboration : Time (s): cpu = 00:00:09 ; elapsed = 00:00:11 . Memory (MB): peak = 1310.566 ; gain = 138.230 ; free physical = 552 ; free virtual = 6460
---------------------------------------------------------------------------------
RTL Elaboration failed
INFO: [Common 17-83] Releasing license: Synthesis
276 Infos, 5 Warnings, 0 Critical Warnings and 4 Errors encountered.
synth_design failed
ERROR: [Common 17-69] Command failed: Synthesis failed - please see the console or run log file for details
INFO: [Common 17-206] Exiting Vivado at Tue Jul  3 21:18:42 2018...
make: *** [prj/mercury/out/red_pitaya.bit] Error 1

RadimPav, 2018-09-21:
**I have followed the tutorial : https://red-pitaya-fpga-examples.readthedocs.io/en/latest/_downloads/StartprogrammingFPGAusingRedPitayaboard.pdf** 

**And the result of my compilation is following (bit-stream is not generated) :/** 

INFO: [Synth 8-256] done synthesizing module 'axi4_slave' (6#1) [/home/redpitaya/GitHub_RedPitaya/RedPitaya/fpga/rtl/axi4_slave.sv:49]
ERROR: [Synth 8-439] module 'system' not found [/home/redpitaya/GitHub_RedPitaya/RedPitaya/fpga/prj/classic/rtl/red_pitaya_ps.sv:92]
ERROR: [Synth 8-285] failed synthesizing module 'red_pitaya_ps' [/home/redpitaya/GitHub_RedPitaya/RedPitaya/fpga/prj/classic/rtl/red_pitaya_ps.sv:28]
ERROR: [Synth 8-285] failed synthesizing module 'red_pitaya_top' [/home/redpitaya/GitHub_RedPitaya/RedPitaya/fpga/prj/classic/rtl/red_pitaya_top.sv:52]

Finished RTL Elaboration : Time (s): cpu = 00:00:06 ; elapsed = 00:00:06 . Memory (MB): peak = 1107.039 ; gain = 169.555 ; free physical = 3889 ; free virtual = 6968

RTL Elaboration failed
INFO: [Common 17-83] Releasing license: Synthesis
106 Infos, 0 Warnings, 0 Critical Warnings and 4 Errors encountered.
synth_design failed
ERROR: [Common 17-69] Command failed: Synthesis failed - please see the console or run log file for details
INFO: [Common 17-206] Exiting Vivado at Fri Sep 21 09:55:33 2018...
Makefile:40: recipe for target 'prj/classic/out/red_pitaya.bit' failed
make: *** [prj/classic/out/red_pitaya.bit] Error 1

**Does anybody know how to advance and how to make him see the module 'system'?**

waqarrashid33, 2019-04-08:
I am following the same tutorial and get the same error.
Did you guys manage to find a way around it or do you know any other tutorial or docs that helps quick start with red pitaya

mike-slackenerny, 2019-10-14:
I had the same issue. The problem is related to the Vivado version. To move forward, you need to upgrade the scripts prj/XXX/ip/systemZ10.tcl
I found a post that recommended to update them using the GUI and Vivado [https://forum.redpitaya.com/viewtopic.php?t=1654]; however, it would not work for me as Vivado was complaining about missing IP, so I switched to editing the files. You need to change the line that sets the vivado version to your installation:

set scripts_vivado_version 2019.1

closed 2023-07-06 by b-swat