RedPitaya/RedPitaya #152: compiling scpi-server: wrong interface for SCPI_Init(...)

KarlZeilhofer, 2018-07-22:
Hi, 

I'm trying to compile the scpi-server as a standallone application for the raspberry pi. 
I've cloned the RedPitaya repo https://github.com/RedPitaya/RedPitaya.git and then cloned into the directory RedPitaya/scpi-server the scpi-parser from https://github.com/RedPitaya/scpi-parser.git

Then the folder structure is as documented in the README.md
[image]

After fixing an initialization error for the `scpi_context.registers` (line 230 in scpi-commands.c) by changing the declaration of `struct _scpi_t`
from 
```
 {
	...
	scpi_reg_val_t registers[SCPI_REG_COUNT];
	...
};
```
to
```
 {
	...
	scpi_reg_val_t* registers;
	...
};
```
I'm stuck at another rather big problem:
`scpi-server.c` calls in line 234 `SCPI_Init(&scpi_context);`, but the prototype requires many more arguments: 
```
    void SCPI_Init(scpi_t * context,
            const scpi_command_t * commands,
            scpi_interface_t * interface,
            const scpi_unit_def_t * units,
            const char * idn1, const char * idn2, const char * idn3, const char * idn4,
            char * input_buffer, size_t input_buffer_length,
            scpi_error_t * error_queue_data, int16_t error_queue_size);
```

I think there is something mixed up with different repo versions of scpi-server and scpi-parser. 

Can anyone help me, how to get this piece of code compiled?

Thanks, Karl

KarlZeilhofer, 2018-07-22:
Found the change in May 2015 (!!)
[image]

Why is this inconsistency still there in RedPitaya?

b-swat, 2023-07-07:
Code updated for latest versions.
Incompatibility fixed.

closed 2023-07-07 by b-swat