RedPitaya/RedPitaya #160: ddrdump.bit source code

BrantRoom3327, 2019-01-23:
I'm looking to rebuild the Test/rp_remote_acquire/ddrdump.bit firmware.  Where can I get a hold of this code?

Thanks!

JDrozdowicz, 2019-02-22:
Hi, I want to do this as well and I did some reading.
Forum thread regarding this is here: https://forum.redpitaya.com/viewtopic.php?f=7&t=317
Probably you can try to recompile the sources from here: https://github.com/HrRossi/RedPitaya/tree/dev_ddrdump/Test/rp_remote_acquire
(also look at this post: https://forum.redpitaya.com/viewtopic.php?f=7&t=317&start=240#p6696 and this post: https://forum.redpitaya.com/viewtopic.php?f=7&t=317&start=90#p2882 )

otobrzo, 2019-02-25:
Hello everynone,

is there any tutorial how to deal with red pitaya ecosystems? ddrdump.bit is for all versions of RP ecosystem? Or as discussed by  Nils Roos  at  https://forum.redpitaya.com/viewtopic.php?f=7&t=317
it is only for 0.92 version?
And other ecosystem needs to be somehow amended. What does 'boot loader replacement' mean? Just to replace u-boot.scr devicetree.dtb boot.bin from other ecosystem version?

Other question. the C code for rp_remote_acquire needs to be compiled and afterwards run  at red pitaya or at my computer?

Sorry I ma new in RP community and I am totally confused.

Thanks!

JDrozdowicz, 2019-02-26:
Hi,
AFAIK there's no tutorial besides already mentioned forum thread. The easiest way to start with ddrdump is:

(quote from: https://forum.redpitaya.com/viewtopic.php?t=317&start=220 )
> 1) Make new sdcard with last stable release of OS
> 2) place u-boot.scr, devicetree.dtb, boot.bin in root of sdcard - replace files ( eco94.zip)
> 3) create 'tmp' folder in root of sdcard
> 4) place rp_remote_acquire, fddrdump.bit in 'tmp' folder (rp_remote_acquire.zip)
> 5) insert sdcard in RP and boot
> 6) connect to RP
> 7) in case you did not unpack the rp_remote_acquire.zip on a linux system, you will have to change the executable flags of the program with the following commands (the first time only)
> cd /opt/redpitaya/tmp
> rw
> chmod 755 rp_remote_acquire
> 8) in the directory /opt/redpitaya/tmp , execute
> ./rp_remote_acquire
> 
> This will give you a short explanation of the parameters.

rp_remote acquire must be run on RP. Remember to load FPGA configuration file ddrdump.bit

otobrzo, 2019-02-26:
Hi,

I tested the fast aquisition using these steps

1. copy the Test/rp_remote_acquire/ddrdump.bit to the root directory of my RP (ecosystem 0.98)
2. copy the source of rp_remote_acquire and compile it using make command
3. run cat ddrdump.bit > /dev/xdevcfg
4. run ./rp_remote_acquire -m 2 -k 0 -c 0 -d 0 to start streaming server
5. run python script on my computer:

`import socket
import numpy as np
import matplotlib.pyplot as plt

client = socket.socket(socket.AF_INET,socket.SOCK_STREAM)

ip=socket.gethostbyname("172.17.170.45") # IP of redpitaya in server mode: 

# run cat ddrdump.bit > /dev/xdevcfg
#compiled and run on redpitay  ./rp_remote_acquire -m 2 -k 0 -c 0 -d 0

port=14000 # default port for TCP 

address=(ip,port)
client.connect(address)  

Nl = 10000

#while True:
for x in range(0, Nl):
#    print("test1")
    bytes_data = client.recv(1024) # set the amount data transferred
    
    if x == 0: 
        data = np.frombuffer(bytes_data, dtype=np.int16) # from 16bit data to int16 
        data = np.array(data, dtype=float)
        data_all = data
    else: 
        data = np.frombuffer(bytes_data, dtype=np.int16) # from 16bit data to int16 
        data = np.array(data, dtype=float)  
        data_all= np.hstack((data_all,data)) 
        
        

#%%
FPS = 125e6        
time = np.arange(0,np.size(data_all))/FPS
    
plt.plot(time,data_all)`

otobrzo, 2019-02-27:
Hi again,
I also tested the data streaming from both channels.

In this case you need to use RP as client and your computer as TCP server:

1. I start TCP server on my computer using this python script:

```
#%%
import socket
import numpy as np
import matplotlib.pyplot as plt

#%%
TCP_IP = 'MY.COMPUTER.IP.ADDRESS'
TCP_PORT = 14000
BUFFER_SIZE = 1024 

s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.bind((TCP_IP, TCP_PORT))
s.listen(1)
```

2. And then I run on RP ./rp_remote_acquire -m 1 -a MY.COMPUTER.IP.ADDRESS -p 14000 -k 0 -c 2 -d 0

3. In python 
```
conn, addr = s.accept()
print('Connection address:', addr)
```

I got: Connection address: ('RP. IP. ADD.RESS', 53410)

The second number is I guess TCP port which should be 14000 and it varied from time to time
Very often I got error messages on RP side:

> connect failed, Connection refused
> main: problem opening connection.
> connect failed, Connection timed out
> main: problem opening connection.
> connect failed, Connection timed out
> main: problem opening connection.

4. In python
```
#%%
while 1:
    data = conn.recv(BUFFER_SIZE)
    if not data: break
    print("received data:", data)
    conn.send(data)  # echo
conn.close()
```

I got no data from RP.

Any idea?

b-swat, 2023-07-06:
Data capture is now implemented in the streaming application.

closed 2023-07-06 by b-swat