RedPitaya/RedPitaya #163: Asymmetric Multi Processing on RedPitaya

waqarrashid33, 2019-04-17:
Hello Friends,

I am working on a project where I would like to do Asymmetric Multi Processing on redpitaya. Linux on one core and BareMetal/RTOS on other. I am unable to create an account on redpitaya forums so I will try to update my progress here or ask for help here.

Cheers.
Waqar

waqarrashid33, 2019-04-20:
Does OS that is provided by RedPitaya supports openAMP?

pavel-demin, 2019-04-23:
Hi Waqar,

This question was once discussed on the Red Pitaya forum:
https://forum.redpitaya.com/viewtopic.php?f=14&t=560

I even did tests and managed to run Linux on one core and a bare metal application on other. I have some instructions at [this link](https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/bare_metal_test) but they are very outdated.

So, it's technically possible but requires patching Linux and building an additional module.

Cheers,

Pavel

waqarrashid33, 2019-04-24:
Hello Pavel,
I was hoping that I would come across you at some point. Your work would be the base of my work so hats off to that.

I will check the links you provided and will post my questions here. I am unable to login/register to the red pitaya forum, which is quite annoying. Anyway great to have you here.

Cheers,
Waqar

waqarrashid33, 2019-05-01:
How to configure redpitaya kernel to enable remoteproc and rpmsg? Like done here [ug1186](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2017_1/ug1186-zynq-openamp-gsg.pdf)

waqarrashid33, 2019-05-05:
So I managed to compile the kernel using the pull request https://github.com/RedPitaya/RedPitaya/pull/87

Now I am looking into adding the remoteproc and sharedmemory nodes to the device tree, here are the  [bindings](https://github.com/Xilinx/linux-xlnx/blob/master/Documentation/devicetree/bindings/remoteproc/zynq_remoteproc.txt) (describes structure of node for dts).

I am not really sure as to where to add them. I made a few attempts but failed. I think I have to change them in the `RedPitaya/build/fpga/logic/dts`.

Any tips?

waqarrashid33, 2019-08-15:
In order for openAMP to work, we will first need to move kernel to a higher address as Xilinx baremetal and freeRTOS applications only work in lower addresses (Correct me if I am wrong as I read it somewhere and didn't have time to verify it).  As can be seen in the figure below First comes the baremetal application/FreeRTOS, then the shared memory region (done in remoteproc device tree?) and then Linux Kernel.

[image]

So we have to first move the kernel to memory address 0x10000000 (I got this address from ug978 Xilinx doc), to do that see #178 .

After that we will need to modify the device tree and add the remoteproc entry. Its same as done for zedboard and I did it in my youtube video  [Tutorial 07 Asymmetric Multi-Processing on ZedBoard (OpenAMP, remoteproc, petalinux) ](https://youtu.be/k7nQ84Bc08E?t=373)

You also will need to compile the device tree and for that I recommend having a look at the Makefile.x86, specifically the target $(DEVICETREE) and hopefully you can figure it out from there how to compile the device tree sources.

With that setup, you can load firmware to remote processor (CPU1), start and stop it. For verification you can use `cat /proc/cpuinfo` to see how many processors you have running.

For making your own remote application, you can start with Xilinx SDK and follow the OpenAMP Framework for Zynq Devices: Getting Started Guide (UG1186) chapter 2. I will also make a video tutorial about it and will upload it my youtube channel and will add the link here.

I am still figuring out how the linux side application of the AMP system works and whether I can develop an application in userspace. There is no `rpmsg_user_dev_driver` on the linux running on redpitaya and I don't know its purpose and where can I find its source code. In some recent documentations I saw that they don't use this module and that would be a better way if somehow they moved the kernel space code of this module to user space but thats still unclear.

Image Source: https://www.xilinx.com/support/documentation/sw_manuals/petalinux2014_2/ug978-petalinux-zynq-amp.pdf

waqarrashid33, 2019-08-21:
According to the source code of [rpmsg_user_dev_driver.c](https://github.com/Xilinx/meta-openamp/blob/4ba8848df9053106de6bd769d9f7f64f9c6dd260/recipes-kernel/rpmsg-user-module/files/rpmsg_user_dev_driver.c) its description is 

> Sample driver to exposes rpmsg svcs to userspace via a char device

And this has to be compiled if we want to use the examples from Xilinx e.g echo_test etc. There are other types of examples which are all in kernel space. So clarify, the Linux application can either be in Kernel space or partly in userspace and partly in kernel space (e.g rpmsg_user_dev_driver).

waqarrashid33, 2019-08-21:
Now We need to compile both kernel module and userspace application. I don't know yet If we would need to compile the libmetal for it.
I tried compiling libmetal on redpitaya but it was giving error related to the ELF headers of libcrypto.so.1.0.0. I don't even know if this is an elf file but had to leave it as I have so much other stuff to figure out until I need to figure this mess out.

waqarrashid33, 2019-08-26:
Other applications like wget etc are also not working so that means the root file was not created properly.. will have to recreate the root file system and see if this problem persists.. also the compilation can be move into the schroot scripts which are used for creation of root file system.

waqarrashid33, 2019-08-28:
Ok So I managed to run one AMP application and would update here soon.

waqarrashid33, 2019-08-28:
Useful talks that somehow wasn't showing up in google search results..
https://connect.linaro.org/resources/bkk19/bkk19-204/
https://connect.linaro.org/resources/bkk19/bkk19-207/

waqarrashid33, 2019-11-08:
Sorry for being late in updates. Here is part 1 of the video tutorial where I did the Asymmetric MultiProcessing on RedPitaya
[image]

victorK8, 2020-09-04:
Hi Waqar,

I follow your youtube tutorial videos for AMP on RedPitaya, but I have some problems loading kernel modules on OS (End of first video). Did you have some issue with that?

Thanks!!

victorK8, 2020-09-24:
Hi Waqar, 

After running command modprobe zynq_remoteproc, /sys/class/remoteproc folder is empty. Do yoy know about this error ??

Thank you, so much

waqarrashid33, 2020-09-24:
Hello victor,
I am not so sure but it might be a problem with the device tree.

Best regards,
Waqar Rashid

victorK8, 2020-09-24:
Thank you!! 

Best regards!

victorK8, 2020-09-24:
Hi! 

I change your added system-top.dts lines (first video of tutorial) with:

	reserved-memory {
		#address-cells = <1>;
		#size-cells = <1>;
		ranges;
		rproc_0_reserved: rproc@3e000000 {
			no-map;
			reg = <0x3e000000 0x400000>;
		};
		rproc_0_dma: rproc@3e800000 {
			no-map;
			compatible = "shared-dma-pool";
			reg = <0x3e800000 0x100000>;
		};
	};

	zynq_remoteproc@0 {
		compatible = "xlnx,zynq_remoteproc";
		vring0 = <15>;
		vring1 = <14>;
		memory-region = <&rproc_0_reserved>, <&rproc_0_dma>;
	};

from https://github.com/Xilinx/linux-xlnx/blob/master/Documentation/devicetree/bindings/remoteproc/zynq_remoteproc.txt.

Maybe it's interesting for you. Now, it seems to work

Thanks Waqar

victorK8, 2020-09-24:
Running a bare-metal app on cpu1 by using remoteproc modules, i have the next errors:

[  205.874292] remoteproc remoteproc0: bad phdr da 0x3e000000 mem 0x1a488
[  205.880350] remoteproc remoteproc0: Failed to load program segments: -22
[  205.887598] remoteproc remoteproc0: Boot failed: -22

Does anyone know about it? I check some pages that say it's about  devicetree.dtb register. Now, i'm using devicetree.dtb file like waqar in tutorial. My linux version is 2017.2.

Thanks!! So much

victorK8, 2020-09-28:
Finally, by addind next line to devicetree.dtb, it works.

amba {
	elf_ddr_0: ddr@0 {
		compatible = "mmio-sram";
		reg = <0x3e000000 0x400000>;
	};
};
remoteproc0: remoteproc@0 {
	compatible = "xlnx,zynq_remoteproc";
	firmware = "firmware";
	vring0 = <15>;
	vring1 = <14>;
	srams = <&elf_ddr_0>;

};