RedPitaya/RedPitaya #164: Problem Connecting red pitaya to internet

ssooffiiaannee, 2019-04-18:
I wired Red pitaya directly to my PC. I load the Red Pitaya OS of the Red Pitaya website, I can access it from the web browser, but when I try to (for example) access the application marketplace, this message appears :

"RedPitaya is offline and cannot access the server! Please follow quick start instructions to put it online.", I went through the quick start instructions, but I cannot find anything that can help me connect the RP to internet.

waqarrashid33, 2019-05-07:
You seem to have the same problem as I am having #165 
Me too can access it from my browser but can't make it to connect to server.
Can you ping external IPs from the ssh terminal?

SokolovaIrina, 2020-05-08:
Hello @ssooffiiaannee, please let us know if the issue is still actual on your side. Have you tried to implement any actions offered by waqarrashid33? 

By now we haven't been able to reproduce the issue you describes on our end. Is it possible to provide us with some video or screenshots? In case we miss anything and can have a full image with the help of screenshots/video. Thank you in advance.

fossum, 2022-01-12:
I think I might have the same issue. The web interface seems fine and I can ping google.com from the terminal, but I cannot use the OS Update.

Side note: The link for help is broken. Currently it points to http://redpitaya.com/quick-start/
Version: STEMlab_125-xx_OS_1.04-9_stable
[image]
```
root@rp-f008ae:~# ping google.com
PING google.com (142.251.33.110) 56(84) bytes of data.
64 bytes from sea30s10-in-f14.1e100.net (142.251.33.110): icmp_seq=1 ttl=118 time=14.4 ms
64 bytes from sea30s10-in-f14.1e100.net (142.251.33.110): icmp_seq=2 ttl=118 time=14.3 ms
^C
--- google.com ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1001ms
rtt min/avg/max/mdev = 14.313/14.365/14.418/0.130 ms
root@rp-f008ae:~#
```

ssooffiiaannee, 2022-01-12:
Hello, The problem with this is that I wired the red pitaya directly to my PC, I wired it to my router and it worked.

fossum, 2022-01-12:
Might not be a bad idea to close your issue for others.

wa2zkd, 2022-01-22:
> I think I might have the same issue. The web interface seems fine and I can ping google.com from the terminal, but I cannot use the OS Update.
>
> Side note: The link for help is broken. Currently it [...]

Is the bad link thing real or will it be fixed

closed 2022-01-22 by ssooffiiaannee

parkerlreed, 2022-01-23:
This is wired to router and still doesn't work.

[image]

[image]

```
root@rp-f04ab3:~# ping google.com
PING google.com (142.250.68.14) 56(84) bytes of data.
64 bytes from lax17s44-in-f14.1e100.net (142.250.68.14): icmp_seq=1 ttl=114 time=77.1 ms
64 bytes from lax17s44-in-f14.1e100.net (142.250.68.14): icmp_seq=2 ttl=114 time=73.4 ms
64 bytes from lax17s44-in-f14.1e100.net (142.250.68.14): icmp_seq=3 ttl=114 time=73.8 ms
^C
--- google.com ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2001ms
rtt min/avg/max/mdev = 73.422/74.808/77.193/1.693 ms
```

parkerlreed, 2022-01-23:
And yeah the getting started link is dead

fossum, 2022-01-24:
@parkerlreed, did you manually upgrade to the latest firmware? That fixed my update issue.

parkerlreed, 2022-02-15:
@fossum link?

crtvalentincic, 2022-02-15:
The latest OS releases can be downloaded from here: https://redpitaya.readthedocs.io/en/latest/quickStart/SDcard/SDcard.html#upgrading-red-pitaya-software

parkerlreed, 2022-02-23:
Thanks. That is indeed working. Now to regrab everything :)