RedPitaya/RedPitaya #165: dns problems with newly compiled linux for redpitaya

waqarrashid33, 2019-05-07:
I compiled linux for the redpitaya using the instructions given in [ecosystem](https://redpitaya.readthedocs.io/en/latest/developerGuide/ecosystem/ecosystem.html). Now when I boot its unable to resolve dns queries. `sytemd-resolve` does work.

In the systemd-network configuration files for the wired interface the option `dhcp=yes` about which I am not sure if its takes the dns sever ip from the dhcp server.

What can be issue here? I am been a user of Linux but am quite new to development/compilation of Linux.

`systemd-resolve www.google.com`    PASS
`ping www.google.com`     FAIL

waqarrashid33, 2019-05-07:
The issue was solved when I check for the file `/etc/resolv.conf` which I am not supposed to touch manually but this file wasn't existing in my redpitaya linux. I created the file and added the following to it:
```
nameserver 127.0.0.53
nameserver 8.8.8.8
```

I will continue working on it and will update more if I found something else.

closed 2019-05-07 by waqarrashid33

waqarrashid33, 2019-05-15:
image.sh will run network.sh which will configure the networking.

parkerlreed, 2019-07-30:
I'm facing the same issue but the resolv.conf is there and correct 

[image]

I can ping out to the Internet when SSH'd into the board

benbarrowes, 2020-02-28:
I have the same issue as parkerlreed. My RP is unable to connect to the update server after getting the same two green checkmarks.

My wifi dongle (the recommended one) is recognized, but does not seem to be able to connect to any network. It tries, shows the right SSID for a second, then goes back to "none".

Right now my connection method is direct ethernet to my PC. To get internet capability to the RP, I had the share my incoming internet connection (to the PC) with "other users" from the connection in the control panel. Even though this seems to work, I still can't connect to the RP server.

OS is 0.99-43 STEM16.

parkerlreed, 2020-02-28:
I think I solved this by grabbing the latest OS manually. After that the update checks worked as expected. I think the server location at one point changed.

GGGGGx, 2022-01-24:
I'm facing the same issue.Has anyone solved it？
[image]

b-swat, 2022-01-24:
https://github.com/RedPitaya/RedPitaya/issues/226