RedPitaya/RedPitaya #167: Question: Which make file is called inside the chroot?

waqarrashid33, 2019-05-08:
Which make file is invoked by this?

Looks like its the one in the main RedPitaya directory but how can it be called from the chroot environment? isn't it outside of it?
```
schroot -c red-pitaya-ubuntu <<- EOL_CHROOT
make
EOL_CHROOT
```

jim-martz, 2019-05-22:
Hello,

I am stuck here at this too.
I chroot into the RP ARM environment, but no RedPitaya director or Makefile to run.
I think maybe Qemu is used to access/run the Makefile in the RedPitaya outside the chroot,
but I am not sure how that works. It is not working for me.
Have you gotten this figured out, or got it to work yet?
Any helpful info?

Thanks,
Jim

waqarrashid33, 2019-05-22:
It works for me. I just don't understand it much.
What kind of error are you getting?

jim-martz, 2019-05-22:
This is the error I get when the above schroot command is run:
W: Failed to change to directory ‘/RedPitaya’: No such file or directory
I: The directory does not exist inside the chroot.  Use the --directory option to run the command in a different directory.
W: Falling back to directory ‘/root’
make: *** No targets specified and no makefile found.  Stop.

And I looked, there is not /RedPitaya directory or Makefile in the
ARM environment that is created in the chroot. The rest of the environment 
seems to be there though.  I do not know how  /RedPitaya or the Makefile
get into the ARM environment. I just know they are not there for me.

The Schroot is using this file for the environment:
redpitaya_ubuntu_13-14-23_25-sep-2017.tar.gz
There is not  /RedPitaya directory in there.

Thanks,
Jim

jim-martz, 2019-05-22:
Ok, I answered one of my questions.
After looking everywhere trying to figure out what is trying
to change the director to /RedPitaya after the environment is loaded.
It turns out that it simply looks like schroot is just trying to change
to the directory that the schroot was run it, and it was RedPitaya that
I was running it from. If I run the schroot from another director, it will try 
to change to that different directory when the environment loads.

But what I have not figured out is where or how it will the get the Makefile.
There is no Makefile in redpitaya_ubuntu_13-14-23_25-sep-2017.tar.gz
And I don't know how its to run it from the host RedPitay directory once it is 
in chroot jail. I am tempted to just copy the Makefile over into the chroot environment
and see what happens. Did you have to copy the Makefile over to get it to work?

Thanks,
Jim

waqarrashid33, 2019-05-23:
No I didn't have to. I will look into my notes later to see if I can figure
out the problem.

jim-martz, 2019-05-23:
Ok, I think I got it figured out...
Schroot by default(and desktop profile) will mount the /home directory
in the chroot environment. So, if the RP ecosystem is in /home/RedPitaya
for example, then when the Schroot commend is run from /home/RedPitaya
the RedPitaya directory will be in the chroot environment(along with the Makefile)
and the schoot command as shown in the RP documentation will work fine.
But I did not put the RedPitaya in the /home director... so that is why it could
not change to that directory when the chroot environment loaded, and that
is why I had no Makefile there.

In conclusion for someone else that has a similar problem, make sure
you put the RedPitaya ecosystem in the /home directory...

Thanks,
Jim

waqarrashid33, 2019-05-23:
Good that you solved it.
Did you manage to create an account on redpitaya forums? I am been trying to make an account from more a month and no one is answering. I feel like I made a big mistake choosing this board.

jim-martz, 2019-05-24:
I don't think I even tried to make an account on the redpitaya forums. I think there is more activity
on the github forum, so I posted on github.
I have contacted them through there support email, but that was more asking about the possibility
of writing some code as "a service" if I am not able to quick enough. And I did get replies to me email. 
Honestly I never did expect much free support from RP directly. For the low cost of the board you
are just paying for the hardware. If NI sold a similar board (eg. SBRIO) it would cost over $1,000USD.
Just like anything in life, you get what you pay for, and don't get what you don't pay for.
So far, I feel that I got what I payed for with the RedPitaya, no regrets yet.

Thanks,
Jim

closed 2023-07-10 by b-swat