RedPitaya/RedPitaya #171: Ecosystem build failed, no .bit file to copy

jim-martz, 2019-05-15:
Hello,

I am trying to build the ecosystem following these instructions:
https://redpitaya.readthedocs.io/en/latest/developerGuide/ecosystem/ecosystem.html

I get to the point of building the Makefile.x86 but it fails here:
cp build/fpga/logic/fpga.bit    build/fpga/fpga_la.bit
cp: cannot stat 'build/fpga/logic/fpga.bit': No such file or directory
Makefile.x86:95: recipe for target 'build/fpga' failed
make: *** [build/fpga] Error 1

If I look in build/fpga/logic/ there is no fpga.bit file, it is fpga.dtbo
And there is a note in the script " TODO, should be removed at next OS release".
Is there something that has changed and I need an updated Makefile.x86?

My build machine is Debian (Stretch), Vivado 2018.3
I will try commenting out the 3 lines below the "remove" comment that
copy the .bit files, but will that cause problems later on?

Thanks,
Jim

waqarrashid33, 2019-05-15:
I think you will need vivadi 2017.2. Its somewhere in the docs.

jim-martz, 2019-05-15:
Well, I switched to Vivado 2017.2 and still get no .bit file, only .dtbo
This time it actually took about 40 minutes to build the fpga stuff, so it
looks like Vivado needs to be version 2017.2
If I comment out the lines as described before I don't get any further.
Because there is no dts file/directory to copy either.
Any idea what is going on here?
I did do a "make -f Makefile.x86 clean" before I tried 
to make it with version 2017.2

Thanks,
Jim

jim-martz, 2019-05-16:
Still no luck today.
I tried deleting the whole RedPitaya directory and starting over.
I followed the "Ecosystem Guide" exactly and I still get the same error.

I do know a little more about what is not working.
It looks like the classic, v0.94, and axi4lite projects build.
But logic and mercury are failing: 
 "Makefile:41: recipe for target 'prj/logic/out/red_pitaya.bit' failed"
I dig through the screen output, but I can not figure out why,
but I do not know much about Vivado at this point.
I did see this warning: "WARNING: [Board 49-91] Board repository 
path 'brd' does not exist, it will not be used to search board files."
But that warning is on the other projects that do build.
I am not sure what the warning means, I do see the board file
in fpga/brd/redpitaya/1.1/
I will try to attach the screen output in a text file,
can someone look at it and tell me what is going wrong?

Thanks,
Jim
[RPmake.txt](https://github.com/RedPitaya/RedPitaya/files/3188872/RPmake.txt)

waqarrashid33, 2019-05-16:
I will try to look into it tomorrow. Currently I am stuck in making
remoteproc work.

jim-martz, 2019-05-17:
I made a little progress today.
I tried to build the projects individually.
The result was the same, but I got a little more info.
When building logic it failed with this:
/opt/Xilinx2017/Vivado/2017.2/bin/loader: line 179:  5228 Killed                  "$RDI_PROG" "$@"
make: *** [prj/logic/out/red_pitaya.bit] Error 137
Makefile:41: recipe for target 'prj/logic/out/red_pitaya.bit' failed
Parent process (pid 5228) has died. This helper process will now exit

I got passed this by changing the maximum threads by adding
set_param general.maxThreads 1
to red_pitaya_vivado_Z10.tcl

The build went further, but failed with an error similar to what I am
getting with the mercury project:
ERROR: [Common 17-39] 'synth_design' failed due to earlier errors.
Although I am not finding the earlier errors off hand.
Will look more next week...

Thanks,
Jim

jim-martz, 2019-05-20:
Ok, I got it all to build, finally!
There was an error listed in the vivado.log:
ERROR: [Common 17-179] Fork failed: Cannot allocate memory
The fix was to increase the memory available to 4Gb in the VM.

So, both problems stemmed from using the VM VirtualBox.
In summery for those trying to build the ecosystem in VM VirtualBox.
1. Make sure you install the same version of Vivado that the ecosystem
   was made in. (would apply to any setup)
2. Add set_param general.maxThreads 1 to red_pitaya_vivado_Z10.tcl
    by default VirtualBox is set to allows only 1 possessor.
3. Make sure the allowed RAM for the VM you setup is at least 4Gb.
    This is also a VirtualBox setting.

Thanks,
Jim

closed 2019-05-20 by jim-martz