RedPitaya/RedPitaya #175: Added Additional Decimation

jim-martz, 2019-06-20:
All,

I have added the flowing decimation to the RP code:
1, 2, 4, 8, 16, 32, 64, 128, 256, 512, 1024, 2048, 4096, 8192, 16384, 32768, 65536
This was added to the Acquire command line utility and the SCPI Server.

I don't know the Github thing that much, so I will just attach a zip
file with the modified files. 

Thanks,
Jim
[RP_Decimation.zip](https://github.com/RedPitaya/RedPitaya/files/3311570/RP_Decimation.zip)

Sophia-Gi, 2021-02-03:
Hi Jim,

thank you for the files! I just found your code and right now I'm trying to get it to work for my project. I edited the eight files you mentioned in you zip via SSH on my RedPitaya but the new decimation factors don't work yet. The signals I acquire with a new factor don't look right anymore. I'm new to working with the files on the RedPitaya so I don't know if there are any more settings/files I need to change to get everything going. Can you help me out?

Thank you
Sophia

b-swat, 2023-07-06:
The current version of the SCPI server supports such decimations.

closed 2023-07-06 by b-swat