RedPitaya/RedPitaya #178: How to Change Kernel base address?

waqarrashid33, 2019-07-27:
Hello I am trying to run the RedPitaya in AMP mode and for that I need to change kernel base address in order to make room for FreeRTOS/Baremetal to run. How do I do that?

For some architectures I found that in the menuconfig they have PHYSICAL_START but its not available for zynq devices. Any Idea?

waqarrashid33, 2019-07-28:
So according to the config files, the kernel is loadable to other memory addresses. The part responsible for that is (Makefile.x86):

```
$(LINUX): $(LINUX_CONFIG)
	make -C $< ARCH=arm CFLAGS=$(LINUX_CFLAGS) -j $(shell grep -c ^processor /proc/cpuinfo) UIMAGE_LOADADDR=0x8000 uImage

```

I guess I will have to changes in this line to understand it better.

ylectron, 2019-08-07:
Hey Rashid,
We noticed your awesome work! and we will be more than happy to discuss some future collaborations.
  Please drop me a message at: support@redpitaya.com

Kind regards,
Nick

waqarrashid33, 2019-08-12:
Anyone knows how to verify that the kernel is loaded into the specified address above?

waqarrashid33, 2019-08-13:
For verification, one way is to see output of u-boot when its loading the kernel. 

```
Booting Linux kernel with ramdisk and devicetree
## Booting kernel from Legacy Image at 02004000 ...
   Image Name:   Linux-4.9.0-xilinx
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4670720 Bytes = 4.5 MiB
   Load Address: 10000000
   Entry Point:  10000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 04000000
   Booting using the fdt blob at 0x4000000
   Loading Kernel Image ... OK
   Loading Device Tree to 1cb3d000, end 1cb45074 ... OK

Starting kernel ...
```

I am still not sure if there is way to find it at runtime from linux kernel.

waqarrashid33, 2019-08-15:
From some further tests, I wasn't able to verify the kernel load-address/entry-point at runtime but after adding the remoteproc into the device tree and making a remote firmware (bare metal application that would be executed on CPU1) using Xilinx SDK and then loading it using remoteproc and running it, I didn't get any errors which kind of suggests that the kernel is where it should be for for AMP applications. I will update this comment with a video link that I am working on.

closed 2019-08-27 by waqarrashid33