RedPitaya/RedPitaya #180: How to download kernel headers for redpitaya?

waqarrashid33, 2019-08-22:
```
root@rp-f05d48:/# uname -r
4.4.0-xilinx

root@rp-f05d48:/# apt-get install linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-headers-4.4.0-xilinx
E: Couldn't find any package by glob 'linux-headers-4.4.0-xilinx'
E: Couldn't find any package by regex 'linux-headers-4.4.0-xilinx'
```

Should I download the kernel source of redpitaya from redpitaya linux repo?

LuAPi, 2021-02-02:
It'd be helpful if a .deb for the linux headers was provided. I'm trying to build a driver for a usb-wifi module and it requires the linux headers.

b-swat, 2023-07-06:
Linux headers added to the ecosystem and placed in the include\linux folder

closed 2023-07-06 by b-swat