RedPitaya/RedPitaya #184: Application error：Could not initialize the application with default parameters.

gaohuiguo88, 2019-10-15:
Has anyone met this problem? When I am trying to run "Oscilloscope_lock in+pid" application, the page shows "Application error：Could not initialize the application with default parameters." I am so confused because it seems no reason to have such problems. I just ran this application successfully  a week ago. I reloaded new image file into my Red Pitaya and even changed the computer and web browser but nothing works. Could anyone figure out what causes the situation and the possible solution? 
[image]

zfiw28, 2020-11-10:
Hi, Have you saved the problem yet?   I came across the same problem recently.  So frustrated.  It could come from the wrong OS imag or ecosystem version for Redpitaya.

Lightsaver7, 2025-05-20:
The Lock_in+PID application is not an official Red Pitaya application. If you have issues with it you should contact the application developers. Please note that Lock-in+PID application developed by Marcelo Luda will not work on 2.00 Red Pitaya OS as it needs to be updated first.