RedPitaya/RedPitaya #191: Example code for 'Red-Pitaya SPI test' problem.

uiuatiq, 2019-12-18:
Hello Everyone,
I am trying to modify the c program of SPI test which is found here (https://github.com/RedPitaya/RedPitaya/tree/master/Examples/Communication/C/spi) for my particular operation. My problem is in "Mode Selection", I try to set the mode 0,1,2, or 3 but the clock always stays at a logic high level then it goes logic low when data start to sample. But, the slave device that I am using that expects data clocked/sampled in the rising edge of the clock. This looks opposite and I try to solve that by choosing a different model, unfortunately in every mode I select the clock to remain unchanged(start at logic high). This is wrong here. How can I change the settings?
In this picture, first yellow wave is clock and blue one is data.
[image]

I am stuck at this point. Would anyone like to give me some suggestions on this, please?
Looking forward to hear from you guys. 
Thank you in advance.

uiuatiq, 2020-01-02:
Hello Guys, The mode selection part is now ok
The SPI bus speed control need to manage. In the red pitaya documentation site the spi bus speed set to 1MHz, but I need around 70Khz for the slave device to work on. So, I try to change the spi bus speed but regardless I change the clock speed it does not change in the observation from  Oscilloscope.  The first line is the clock, 2nd is data and third is LE. The clock speed is high here. How can i change the clock speed from the code?

[image]
[image]

[image]

Thank you so much and looking forward to getting some response.

brrgama, 2022-01-31:
Hi! Can you please share the solution for your first question? I am facing the same problem.