RedPitaya/RedPitaya #197: Where can I download an old version of SD card image? I need 0.94

shuow2, 2020-05-24:


b-swat, 2020-05-28:
Hey.
0.94 is this probably a version of the FPGA?
If you said from what date the assembly would be a little easier.
Or look here:
http://downloads.redpitaya.com/downloads/
there are very old builds.

shuow2, 2020-05-28:
Hi,

I am using "impedance analyzer" application. The version information is as following. The assembly date might be 12/21/2015 or so.

[cid:ca4d0074-1e3f-462f-8d09-97fa0974231c]

Thanks so much for your help!

Best,
Lillian

http://downloads.redpitaya.com/downloads/

b-swat, 2020-05-29:
I uploaded old ecosystem images. Maybe you will find the right one.
http://downloads.redpitaya.com/downloads/0.92-0.94%20old%20ecosystems/

shuow2, 2020-05-29:
Hi,

Thanks for your reply. But it seems they cannot be downloaded...Could you please help fix it?

[cid:43818121-6aac-4ae1-969a-dca860b2c5e1]
Thanks,
Lillian

http://downloads.redpitaya.com/downloads/0.92-0.94%20old%20ecosystems/

b-swat, 2020-06-01:
I fixed the permissions.
Check out

closed 2023-07-11 by b-swat